particle-iot/device-os #378: Inconsistent behavior with WiFi.ready() in SYSTEM_MODE(MANUAL)

piettetech, 2015-01-14:
I have found a case where WiFi.ready() will return true when the CC3000 is off.

If you bring up the network in stages -
`WiFi.on();
WiFi.connect();
Spark.connect();`

And if you take the network down in stages -
`Spark.disconnect();
WiFi.disconnect();
WiFi.off();`

Then `WiFi.ready();` works as expected.  

However you should be able to go from a fully connected network w/Cloud and call `WiFi.off();`
When you do `WiFi.ready();` returns TRUE every time you call it after your call to `WiFi.off();`

`                WiFi.off();
                Serial.print("Waiting for WiFi to turn off ");
                while (WiFi.ready()) { Serial.print("-"); Spark.process(); delay(100); }
`

The while loop never exits if you skip the call to `WiFi.disconnect();`

Simple demonstration code can be found [here](https://github.com/piettetech/WiFiTest)

m-mcgowan, 2015-01-14:
Thanks for this - it's a great bug report! :+1:  

The example program is also great to have! (Would be extra awesome if it contained only code directly related to the problem.)

Looking through the source code I see the problem. The fix is to have `Wifi.off()` call `Wifi.disconnect()` internally so that various flags are set to indicate the wifi is disconnected.

piettetech, 2015-01-15:
@m-mcgowan 
Sorry for the extra code, I usually have to hack down another test utility to demonstrate the problem. 

Yes, we need to make sure the flags are set correctly if `WiFi.disconnect()` is skipped when tearing down the WiFi connection.

Thanks!

m-mcgowan, 2015-01-19:
Fix is here. https://github.com/spark/firmware/commit/3414137

I chose to reset the flags rather than call `disconnect()`. The rationale is that disconnect controls if the system automatically connects wifi. So the sequence of events:

```cpp
WiFi.connect();
// ....
WiFi.off();
// ...
WiFi.on();
```

Will result in Wifi being automatically re-connected after `WiFi.on()` is called. If this isn't desired, then adding a call to `WiFi.disconnect()` will prevent automatic reconnection when wifi is switched on.

closed 2015-01-19 by m-mcgowan

piettetech, 2015-01-19:
@m-mcgowan 
Currently when using `SYSTEM_MODE(MANUAL)` the system does not automatically connect when the WiFi is intentionally turned off and then on again, regardless if you call WiFi.disconnect before you turn it off.  What you have described seems to change that behavior?
When using `SYSTEM_MODE(SEMI_AUTOMATIC)` or `SYSTEM_MODE(AUTOMATIC)` then what you describe sounds correct.

nolanhergert, 2015-02-28:
This fixes the problem I was having too! Where is this fix committed to? Is the HAL branch "safe" to use?

nolanhergert, 2015-02-28:
Nevermind, I see that the fix is in the HAL branch and other people recommend using it now. Thanks!