particle-iot/device-os #380: CC3000 stuck with 76.83.0.0 DNS IP Address

piettetech, 2015-01-15:
The TI CC3000 chip can get into a state where the DNS IP address is always 76.83.0.0 and any connect using a hostname will always fail.

I have developed a way to temporarily correct this state and allow the CC3000 pickup the correct DNS IP address from the AP.

I consider this fix temporary because the corrected state does NOT survive power cycles or AP reconnects.  It does allow someone to use the code in their `startup()` to fix a core having problems with the CC3000 DNS IP setting as well as help you troubleshoot the problem.

Since this touches the CC3000 directly I am limiting the availability of this code.

I have spent several days developing this app.  I don't have the deep internal knowledge of the CC3000 to create a patch or fix, so my motives are purely to help smarter people get to the root of the problem.

Please contact me if you would like to try this program.

dmiddlecamp, 2015-01-15:
Does re-running the cc3000 patch fix this issue?

If you use the spark-cli, and place your core in dfu-mode ( https://github.com/spark/spark-cli )
spark flash --usb cc3000

Does that fix it?

piettetech, 2015-01-15:
@dmiddlecamp 
Yes!  I wish I had tried that a few days ago. I remember trying the patch several months ago and It did not correct the problem - so I just put the core on the shelf.  When I went to use the a few days ago I rediscovered the problem and started to troubleshoot.  I didn't think of applying the patch again.  

Now I'm curious, did the patch change in the last 6 months?

dmiddlecamp, 2015-01-15:
Hi @piettetech,

  Yes!  There were one or two new versions in the last 6 months I believe, and if nothing else applying it might help reset the internal state of the cc3000 I suspect.  :) 

Thanks,
David

davids5, 2015-01-15:
In the past, when the CC3300 (Even just the Ti module sans spark) has gotten into a state where the addressing returned looks very messed up . The only way I could recover it was to repatch it.

bkobkobko, 2015-01-15:
Several folks in the forum have reported that re-patching the TI part at least temporarily gets them going again.

I would love to have someone that this happens to capture packets but it has never happened to me and is hard to reproduce.  At least some of the people that get this problem have a "slow" network.

piettetech, 2015-01-15:
I had this problem about 6 months ago and none of the patches at that time would correct it.  I ended up getting replacements.  Then another of my cores a month or so after that failed in the same way and I tried patching it with no success and putting it on the shelf.

@dmiddlecamp  I'm thrilled we now have a fix.
@bkobkobko I have an app that if you find a core with the problem the app will automatically get it working for at least that session and you can capture what ever you want during the process.

I uploaded the firmware (https://github.com/piettetech/DNSTest)

bkobkobko, 2015-01-16:
Hi @piettetech Scott,

It would great if you could report back (and possibly reopen this) if patching the TI part does or does not fix your problem.

Thanks for the tool but first I need a way to make it fail!

piettetech, 2015-01-16:
@bkobkobko 
Yes, the current cc3000 patch cleared the problem.

m-mcgowan, 2015-04-05:
@piettetech Was that the 1.14 driver (service pack 1.32)? If so, feel free to close this issue if that is a workable fix for you.

bkobkobko, 2015-04-05:
I think what @piettetech meant was that re-patching the TI CC3000 (i.e. performing deep-update) clears the fault until next time.  So I don't see this as a fix so much as a work-around.  But it also points the finger at what the real cause is likely to be--something inside the TI part.

piettetech, 2015-04-06:
@m-mcgowan  @bkobkobko stated it correctly, re-patching cleared the fault. I will close the ticket until this occurs again.

closed 2015-04-06 by piettetech