particle-iot/device-os #381: Using REST to upload firmware is causing Spark Core to stay in DFU mode, needs USB reflash afterwards

PombeirP, 2015-01-17:
I'm writing an Android app which prepares a Spark Core with a custom firmware. I am able to send the firmware in text fomat to the Core, and see the purple LED blinking, but after the firmware is flashed, it changes to orange blinking. I had to use dfu-util to reflash the unit. I've confirmed that the source being sent is correct, and the HTTP response says it is updating. The same program can be flashed successfully using the Web IDE.

For reference, the sources are [here](https://github.com/PombeirP/SparkWOL/blob/master/app/src/main/java/com/pedropombeiro/sparkwol/MainActivity.java#L156)

Thanks for any help.

m-mcgowan, 2015-01-19:
Could you describe your upload process in detail - the URLs you are posting to, the files being sent etc.. so that we have the details to be able to retrace your steps and hopefully reproduce the issue. Thanks!

PombeirP, 2015-01-19:
Hi Matt,

The whole thing is available online. The project is at
https://github.com/PombeirP/SparkWOL. This is an Android app which among
other things will propose to flash a firmware if it detects it is not on
the Spark. The firmware file is
https://github.com/PombeirP/SparkWOL/blob/master/app/src/main/res/raw/SparkScript.cpp,
and it is being flashed through a Retrofit call using the interface at
https://github.com/PombeirP/SparkWOL/blob/master/app/src/main/java/com/pedropombeiro/sparkwol/SparkService.java
(`flashFirmware`). The call is made at
https://github.com/PombeirP/SparkWOL/blob/master/app/src/main/java/com/pedropombeiro/sparkwol/MainActivity.java#L189
.

Hope this helps,
Pedro

PombeirP, 2015-01-25:
I think I figured out what was happening. Even though I was uploading with MIME type 'text/plain', it was probably being flashed as if it was a binary. Once I tried with a real binary, it worked perfectly. Sorry about the false alarm. Maybe a safeguard for this situation could be implemented in the web service, to avoid other people going through the same.

closed 2015-01-25 by PombeirP

m-mcgowan, 2015-01-25:
So great you figured this out! Yes, the file you upload should be the compiled binary not the hex, elf or other produced artifacts. I'll defer to my illustrious colleague @dmiddlecamp about checking mime type etc..

PombeirP, 2015-01-25:
I was passing the actual source code to be compiled. not a hex format
though.