particle-iot/device-os #385: WiFi.disconnect() doesn't change visual state

m-mcgowan, 2015-01-24:
The LED states are:

- breathing white: Wifi Off
- breathing blue: Wifi On, Disconnected
- flashing/breathing green: Wifi connecting/connected

Using MANUAL mode and the sequence of calls 

- power on, breathing white
- `WiFi.on()`, breathing blue
- `WiFi.connect()`, flashing/breathing green

But then calling `WiFi.disconnect()` doesn't return to the breathing blue, but remains in breathing green, which is misleading.   (And confusing to weary engineers debugging wlan code.. ;-) )

The fix is to add

```cpp
        WLAN_DHCP = 0;
        LED_SetRGBColor(RGB_COLOR_BLUE);
        LED_On(LED_RGB);
```

To `Wifi.disconnect()`

This change seems to make sense to me.  Anyone disagree?

satishgn, 2015-01-24:
Actually when I tried the above (after OTA flashing the firmware via Web IDE), the behaviour is as expected:

WiFi.connect() => flashing/breathing green.
WiFi.disconnect() => breathing blue.

m-mcgowan, 2015-04-05:
Closing this. Tested as working with latest code.

closed 2015-04-05 by m-mcgowan