particle-iot/device-os #388: Spark.publish() should return a boolean

zsup, 2015-01-28:
Currently `Spark.publish()` returns `void`. However the underlying implementation returns a boolean: `true` when the message is successfully published, `false` when it is not. I'd like to suggest that `Spark.publish()` should also return a boolean: `true` if the message is successfully published, `false` when it is not.

If possible, this boolean should be based not just on whether the message was _sent_ but on whether it was _received_; in other words, `Spark.publish()` should return true when the cloud acknowledges the message. I realize this might not be possible because blocking, but I'll throw it out there in any case.

kennethlimcp, 2015-01-28:
:+1: i think it was mentioned somewhere in the community. `publish()` *should* work like UDP but having a confirmable published message would be awesome though!

m-mcgowan, 2015-02-04:
I see what you mean. The success return code indicates merely that the message was sent - not that it was necessarily received. However, in practice, given that the transport is currently TCP, then sending is as good as receiving in this instance, but that might not always be so if we were to change to a non-connected transport.

m-mcgowan, 2015-02-04:
Closing - this implemented in the feature/hal branch.

closed 2015-02-04 by m-mcgowan