particle-iot/device-os #393: System.FactoryReset() function

kennethlimcp, 2015-02-10:
We have the soft rest exposed using `System.Reset()` but some users are looking for a complete Factory Reset via code.

See: 

https://community.spark.io/t/hard-factory-reset-in-software/9919

m-mcgowan, 2015-02-10:
There is

```
    System.factoryReset();
    System.bootloader();
```

In the wiring API but these are not documented. (I don't know why not.)

`factoryReset()` reboots the device and instructs the bootloader to perform a factory reset.

`bootloader()` reboots the device and enters DFU mode.

kennethlimcp, 2015-02-10:
Can we get them tested and baked into HAL for documentation merged with the photon release?

m-mcgowan, 2015-02-10:
They've been around for a long time. Already tested and working!

closed 2015-02-10 by kennethlimcp