particle-iot/device-os #394: WiFi antenna select on Photon

m-mcgowan, 2015-02-11:
The photon has a built in RF switch to support the built-in antenna or a u.FL antenna.

The system can be set to select automatically which antenna to use, although this may degrade performance as the system continually switches back and forth to determine the best one.

A new wiring api

```
bool WiFi.antennaSelect(AUTO | INTERNAL | EXTERNAL);
```

will allow user code to specifically select the antenna to be used for their application. The method returns `true` if the antenna was successfully switched. (Note that this doesn't detect if the external antenna is connected, only that the system has switched to that antenna input.)

The wiring API boils down to a call to the WWD api - `wwd_wifi_select_antenna()`.

On devices without multiple antennae, the operation is a no-op and always returns `false`.

piettetech, 2015-02-11:
how will this work for products that are based on the P0 that don't have the RF Switch and only a chip antenna?  Can you give any detail on how the `wwd_wifi_select_antenna()` knows only one antenna is available?

I don't have access to the WWD api, is that something that will be made available?  A quick search does not turn anything up on google.

technobly, 2015-02-11:
Hi @piettetech thanks for the great question.  If you would like to have a dedicated u.FL or Chip antenna with the P0, you would just leave pins 12 and 13 of the module unconnected.  These are the RF_SW_CTRLx lines.  In code you would just want to make sure to set some default that is not AUTO, so those RF_SW_CTRLx lines don't toggle.  Even AUTO might settle down after one iteration since there will be a dedicated RF connection.  The lines may toggle once and then lock into whichever one is a stronger signal.  Setting it to INTERNAL or EXTERNAL is a more deterministic coding approach though.

It might be good that we set the default of this method to `WiFi.antennaSelect(INTERNAL);` to cover most bases.  Even if someone created a dedicated u.FL project without an RF switch it would be appropriate (because it doesn't matter how the RF_SW_CTRLx are set in that case, just that they are not toggling).

piettetech, 2015-02-11:
@technobly Excellent!  Sounds like a plan.

technobly, 2015-03-25:
This API has been implemented.

closed 2015-03-25 by technobly

haldean, 2015-06-26:
Hey folks! Trying to use this API in a project of mine; where is this implemented? build.particle.io doesn't says that neither `antennaSelect` or `selectAntenna` exist on the WiFi class.

towynlin, 2015-06-26:
Hey @haldean — current development happens on the develop branch, so you'll find it there as, for example, `WiFi.selectAntenna(ANT_EXTERNAL)`. See this post/thread:
http://community.particle.io/t/switching-between-chip-and-u-lf-antenna/12710/4?u=zachary

haldean, 2015-06-26:
Gotcha. Thanks so much!