particle-iot/device-os #395: rename System.bootloader() function

kennethlimcp, 2015-02-12:
This functions allows the device to enter DFU mode and will not exit even up reset until a firmware is flashed via DFU-util.

However, the name of the function does not fully explain what it does and so on.

m-mcgowan, 2015-02-12:
I can see your point, but also that for most users the bootloader is only visible when it's in DFU mode, so calling it `bootloader` can be a simple way of conveying that. 

I also found persistent DFU mode to be unexpected. Perhaps the bootloader should only enter dfu mode once after the `System.bootloader()` call and on subsequent resets resume normal operation?

(This can be done simply by clearing the DFU mode flag once it's detected and the bootloader enters dfu mode.)

kennethlimcp, 2015-02-12:
I think the issue is: What does `System.bootloader()` even do? My understanding of the behavior is that it simply enters the device into DFU mode.

People usually know bootloader of a piece of code that runs upon boot-up and also manages other stuff in their system..

The term `System.bootloader()` doesn't seem to suggest anything that the user might want to do?

m-mcgowan, 2015-02-12:
Yeah agreed. I'm just siding with the fact there must have been rationale for the existing name, but think that that `System.rebootToDFU()` might be better, although it's more technical it says what it does.  

But this is also a dangerous method - there are classes of devices that cannot be connected to DFU (e.g. no exposed USB port) and would essentially be bricked by this call since DFU mode is persistent, so that's definitely one reason to make DFU mode transient, where a reset resumes normal loading.

technobly, 2015-02-27:
I remember reading this... forgot about it... then was looking through the Docs and came across this functions and instantly made up my mind that it must do something with the Bootloader... but it was all about DFU mode.  Then I remembered this debate :) 

My opinion would be to rename it to `System.dfu()` to keep it simple.  It simply puts the system in dfu mode, much like System.sleep() puts the system to sleep and System.reset() resets the system.

m-mcgowan, 2015-04-05:
A new method `System.dfu()` has been added which causes the core to transiently enter DFU mode - hitting reset will reboot normally.

The existing method `System.bootloader()` is still present, and performs a persistent boot to DFU mode - the core will continue to enter dfu mode even after reset, until new firmware is flashed via DFU. 

My feeling is that a persistent boot to DFU is dangerous, but since it is documented we can make users aware of the risks. I feel there needs to be more emphasis that a binary MUST be flashed via dfu before the device will boot, and contrast that behaviour wth the  `System.dfu()` method.

zsup, 2015-04-05:
I would argue that if it's dangerous, we should deprecate it, even if it's currently available and documented, and provide a clear compiler error.

m-mcgowan, 2015-04-05:
:+1:  to removing.  It's dangerous for devices that don't have a USB connection. But I'd also say the behavior is unexpected - pressing reset should clear any state set by the firmware.  

Can anyone come up with a use case where a persistent DFU mode is necessary?

m-mcgowan, 2015-04-21:
System.bootloader() removed. `System.dfu()` exists and users can optionally pass `true` to make it persistent if that is truly required.

closed 2015-04-21 by m-mcgowan