particle-iot/device-os #400: wireing_wifi has compile type mismatches.

harrisonmetz, 2015-03-05:
in spark_wlan.h 'wlan_profile_index' is of type 'unsigned char' however, wlan_add_profile returns INT32 (long int), and in spark_wiring_wifi.cpp there is the comparison 'if(wlan_profile_index != -1)'

should wlan_profile_index be an int32 or should that comparsion check 0xFF instead of -1? What should the type actually be? and if unsigned char, then the other functions should be changed to match. I'm happy to make the changes, but please let me know the original author's intentions.

m-mcgowan, 2015-03-05:
Hi @harrisonmetz! Thanks for reporting this. It's indeed a type mismatch.

The upcoming release will be based on the code in the `feature/hal` branch, and in this branch we have enabled all warnings as errors for all system code as a key tool to increasing code quality.

That particular piece of code has been factored out of the high level system system code as part of our hardware abstraction - it's been moved to the HAL here.
https://github.com/spark/firmware/blob/feature/hal/hal/src/core-v1/wlan_hal.c#L233

You'll see `wlan_profile_index` has indeed been changed to an `int` to avoid the type mismatch, and in doing so we fix a bug. When `wlan_add_profile()` returns a negative value, this would be cast to the unsigned char value of `0xFF`.  Then in the comparison `wlan_profile_index != -1` it is implicitly converted to an `int`, however, the value will be 0xFF / 255 and not -1, so the comparison will always succeed, even when the profile isn't correctly added.

m-mcgowan, 2015-03-07:
closing since this is fixed in feature/hal.

closed 2015-03-07 by m-mcgowan