particle-iot/device-os #401: System Information / Diagnostics

m-mcgowan, 2015-03-06:
The system has grown in complexity lately, with crc-checked firmware images, monolithic and modular firmware versions, separate user firmware etc..etc...  

Commands to print system info would help testing and provide tools for error diagnostics. The system info could include:

- device ID
- platform ID
- product ID
- bootloader revision
- system firmware type (mono/modular)
- system firmware revision
- factory firmware valid / revision
- (modular) user firmware installed / CRC valid
- (modular) user firmware revision, required system revision

In addition to the firmware details above, the system configuration could also be reported:

- softAP SSID
- public key
- claim code? 
- configured cloud server address
- wifi credentials (excluding password)

The data will be made available over Serial and via a new softAP setup command so it is potentially available for consumption/forwarding via the mobile app.

The system into text would be expressed as JSON for simple generation and parsing, e.g.

```
"system_info": {
"platform_id": 6,
"bootloader_revision": 1,
"firmware_modular": 0,
"firmware_revision": 22,
}
```

If there's data that would be useful to have that I've missed, drop a comment below!

andyw-lala, 2015-03-06:
MAC addr/GUID
Serial # (if different from MAC)
production test date (can be WWYY) & test station ID
For electron, IMEI ?

m-mcgowan, 2015-08-22:
The 's' and 'v' commands are already implemented and cover all but a few of these. Those that remain are available via additional commands, e.g. 'm' for MAC address. Entering SoftAP mode shows the SSID/Packaging Serial #, pressing 'i' shows the Device ID.  

Closing this for now until there is a call to action on this.

closed 2015-08-22 by m-mcgowan