particle-iot/device-os #406: Provide a user API for the 32-bit DWT hardware counter on Photon and Core

technobly, 2015-03-19:
Please reference this community post: https://community.spark.io/t/micros-rolling-over-after-59-652323-seconds-not-71-minutes-solved/3468/18

Working with micros() values can be tricky, since they top out at 59652323 and not UINT_MAX(32bit unsigned long).  It puts more burden on the user to protect against rollover which happens every second (i.e., very frequently)

This super lightweight function would give users access to a more precise counter with full 32-bit resolution, which will allow them to use unsigned integer subtraction to compute time deltas (which automatically protect against rollover).  The user must still be aware this is only valid for time deltas less than ~1 second (59,652,323us).  For very short and precise time measurements, this works perfectly.  For anything larger than 1 second, the millis() counter is a better choice.  Also a combination of the two could yield very precise measurements with intervals greater than 1 second.

m-mcgowan, 2015-03-19:
Yes. I think some inline functions would be perfect here so that the value of the DWT counter can be inspected, and inline so we have no function call overhead.

I don''t think we specifically need to do anything other than provide guidance as to which stm32 platform headers to include and some examples.

technobly, 2015-03-19:
Great!  Is there a foreseeable way to inline it all of the way out to the Wiring library, so no headers are necessary?

m-mcgowan, 2015-03-19:
we could simply add the header to the default application.h includes. Do we need to make a special API? 

If the answer is yes, we could have an inline function like

```
uint32_t SystemClass::ticks() {
    return DWT->CYCCNT;
}

uint32_t SystemClass::ticksPerMicro() {
   return SYSTEM_US_TICKS;
}

```

technobly, 2015-03-19:
Yeah I like that a lot.  ticksPerMicros() would be a clearer name I think.  

Part of me thinks that `System.ticksPerMicros()` is making `SYSTEM_US_TICKS` more complicated... but maybe it's easier to remember this way?

m-mcgowan, 2015-03-19:
cc: @jenesaisdiq.  Richard, we're discussing a potential API to allow users to measure sub-microsecond delays using the system clock counter.

technobly, 2015-03-19:
Not sure if this needs to be a separate github feature request, or if we should combine it here... 

Many like to use the millis() and micros() counters to make timed measurements.  There are some details with using both of these that can be tricky.

A proposal for a simple method for capturing time deltas that would look something like this:

    millisDelta(RESET); // to reset the timed measurement
    int elapsedMs = millisDelta(); // to read the delta from last time it was RESET.
    // if the delta is > unsigned long max, return -1 to indicate overflow.

    microsDelta(RESET); // to reset the timed measurement
    int elapsedUs = microsDelta(); // to read the delta from last time it was RESET.
    // if the delta is > 32-bit/72 for Core or 32-bit/120 for Photon, return -1 to indicate overflow.

So that they are familiar to users of millis(); and micros();

These functions would manage overflow.

andyw-lala, 2015-03-19:
<my_2c>
I don't like that RESET thing very much, I _love_ free running unsigned counters that are large, and so you can just do subtraction to get the delta. It is a deep shame we do not just have a free running  32-bit 1uS counter. That would only wrap after over an hour.

I don't really like the faster-than-1uS timer, even with the divisor available, just because it wraps so fast and needs bludgeoning into usable units.
</my_2c>

technobly, 2015-03-19:
Agreed on all fronts, but the nice thing about that free running 32-bit DWT timer is that it's a hardware timer, not affected by interrupts or anything else.  You still have to poll it, but it's pretty cool.  It also sets a flag when it overflows that we could use to expand it to a 64-bit counter.. but would require some intelligent code to retrieve the correct value (unless there's some built-in atomic way to read 64-bits I don't know about yet).  Again, that might not be as useful for most people that want a super precise measurement that's less than 1 minute long. (oh just realized I was saying 1 second earlier).

Use of the RESET is kind of an easier way to set the start time than creating a separate function, but maybe separate functions would be better... `millisBegin()` and `millisEnd()` ... this is less important for millis() since it does return a full 32 bit unsigned long value, and more so for micros() which does not.  But they would both benefit from error trapping, still though... arguably less benefit for millis() since it wraps very infrequently.

I've already used the DWT timer in many projects, it's been super useful.  It only gets better with the Photon running at 120MHz.  I think that one's a slam dunk to add.  The rest I see value in, but not a high priority by any means.

andyw-lala, 2015-03-19:
I was only talking about hardware timers too. Intend to setup my tick
timers that way, so I can use the higher resolution underlying hardware (eg
by loading the compare  register dynamically in the isr to allow the
underlying timer free run.)

bkobkobko, 2015-03-19:
This needs to have logic added for the wrap-around but wouldn't something like this work?  

```c++
unsigned long micros(void)
{
  static unsigned long microsCounter;
  static unsigned long lastHWCounter;

  unsigned long currentHWCounter = DWT->CYCCNT;  //sample the current time
  unsigned long incMicros =  currentHWCounter - lastHWCounter;
  lastHWCounter = currentHWCounter;

  microsCounter += incMicros/SYSTEM_US_TICKS;
  return microsCounter;
}
```

technobly, 2015-03-20:
Howdy @bkobkobko!  I'm not sure which one of the things above you are trying to address, but I think the way micros() is implemented now will do the same thing without all the extra code.  If I remember correctly it's just `return DWT->CYCCNT/SYSTEM_US_TICKS;`.  The DWT wraps around automatically so there is no need to subtract old from new and accumulate.  

The thing I'm thinking about with something like microsBegin() and microsEnd() (or microsDelta()) would be something that helps manage the wrap around at 59 seconds for you.  It also allows you to essentially set a zero point and get a full 59.whatever seconds before it wraps.  Verses what we have now with micros() where every time you call it could be on the verge of wrapping, or somewhere in between.

bkobkobko, 2015-03-20:
The code I wrote tries to make a counter based on the hardware counter that wraps at 2^32-1. It does this by remembering the micros deltas between calls and adding that to a static var that will roll over after 2^32 so user code does not have to change. Then you don't need anything but micros().

technobly, 2015-03-20:
Ah ok. I see what you were going after now, but the returned value is 32bit divided by 72 so it will never reach its full 32 bit value and wrap as the returned value. You'd need to expand the accumulation to 40 bit or maybe just go all the way for 64 bit before the divide.  I have a feeling the divide is going to start impacting the accuracy of the timing though.

bkobkobko, 2015-03-20:
Hi @technobly 

Only the increment value is scaled by 72 so the final result will reach 2^32-1.

The only limitation is that users need to call this more often than it can wrap, so less than 59 seconds or double that if we can find the overflow flag in the processor spec.

technobly, 2015-03-20:
Wow, I gotta stop looking at stuff late at night.  I see it now.  Sorry for the confusion @bkobkobko ;-)

Yeah needing to call it more often does complicate it.  The idea would be to make it easier to use, intuitive... without strings attached.

m-mcgowan, 2015-04-05:
@bkobkobko  - nice code - I was thinking on the same lines too. If we also add in `millis()` to track potential overflows of the DWT counter then we can build a monotonically increasing `micros()` counter that wraps around at (2^32)-1 (ca. 71 minutes.)

m-mcgowan, 2015-04-05:
> @technobly 
  Yeah I like that a lot. ticksPerMicros() would be a clearer name I think.

Not sure why we need micros in plural - you don't say "miles per gallons". How about we avoid the elision and make the name `ticksPerMicrosecond()`.

> Part of me thinks that `System.ticksPerMicros()` is making SYSTEM_US_TICKS more complicated... but maybe it's easier to remember this way?

The main driver for this is that we are providing isolation of user code from the system implementation. On other cpu architectures the timing details may be radically different - the API allows user code to remain the same yet target different systems.

bkobkobko, 2015-04-05:
I just wanted to say that I think Arduino compatibility should also be a goal, so making the plumbing more complex so the user sees something simple is OK in my book, especially with the HAL abstraction making it look the same everywhere.

m-mcgowan, 2015-04-05:
:+1: definitely! I'm all for handling the complexity in the system to give users an easy life!

m-mcgowan, 2015-08-17:
just a ping that `micros()` has been reworked to have full 32-bit wrap around. Does that change the need for this issue?

closed 2015-09-27 by m-mcgowan