particle-iot/device-os #407: UDP.write should buffer and nothing should be sent until UDP.endPacket

jnm2, 2015-03-20:
`UDP.parsePacket` and `UDP.read` are broken. The Spark will sometimes receive incoming packets `AAA`, `BBBBB` as `AAABBBBB`. I need to be able to rely on the return from parsePacket to tell be the length of the next unread packet.

It's an embarrassment for a serious library to get the basic UDP fundamentals wrong. This will give newbies a skewed idea of UDP and will frustrate people who know what they are doing. Please fix this.

bkobkobko, 2015-03-21:
As I said in the forum, this is an acknowledge by TI fact of life with TI CC3000 on the Spark Core.  Photon will be better.

jnm2, 2015-03-21:
What would it take to fix the UDP class?

jnm2, 2015-03-21:
I would submit a pull request with the fix but I feel somewhat shy. Would that be welcome? I would have a couple questions.

m-mcgowan, 2015-03-21:
Pros are always welcome!

jnm2, 2015-03-21:
Okay! Questions:

 * Should the buffer be statically or dynamically allocated? It'll have to support the MTU size, but it could also be optionally given a lower max in the constructor. For example, my app never sends more than 69 bytes per packet.
 * Is there some kind of error-reporting precedent to be consistent with the rest of the firmware? What should happen when you `write` without first calling `beginPacket`, or try to write too much?

jnm2, 2015-04-03:
Hey I'm new, is there a process to follow? I have those questions before I can help out here.

m-mcgowan, 2015-04-05:
Hi @jnm2! Apologies for the delay - here's some answers for you! :-)

- I suggest having a constructor that takes `uint8_t* buffer, size_t buffer_size` so that the caller can decide how best to allocate the buffer and how large to make it.
- for error handling, use return codes to indicate errors. E.g. having write return 0 when it can't write any more data because the buffer is full. While it's of course good practice to detect caller problems such as a write() before beginPacket() it has to be balanced with cost of detecting those problems.

It would be great if you could provide details of how you test this (e.g. source code for any UDP servers you make), since we also want to test the same on the Photon.

As to process - please see http://spark.github.io/#Contributions

I hope that's enough to get you started, but of course please ask questions as the need arises!

jnm2, 2015-04-11:
Thanks @m-mcgowan. I'm testing against ASP.NET's [UdpClient](https://msdn.microsoft.com/en-us/library/system.net.sockets.udpclient.aspx) on the server side, so I'm not sure what you'd want as far as testing.

m-mcgowan, 2015-04-19:
Testing that sending and receiving respects packet boundaries would be a great start. 

Looking at the [arduino sources](https://code.google.com/p/arduino/source/browse/trunk/libraries/Ethernet/Udp.h?r=1094), there is the `sendPacket()` function that sends an entire packet in one operation without needing `beginPacket()`/`endPacket()`  I would imagine that would avoid confusion and avoids unnecessary buffering when the caller allocates their own buffer, e.g.

```cpp
UDP udp;
char buf[512];

void sendSomething()
{
    // fill the buffer `buf`  
    buf[0] = 0; // etc..
   udp.sendPacket(buf, 512, address, port);
}
```

bkobkobko, 2015-04-19:
In my experience and reading, I do not believe that the TI CC3000 can distinguish received UDP packet boundaries.

I also think in whatever changes are made in this area, Arduino compatibility should be a very high priority.

loopj, 2015-07-05:
Looks like this is finally being fixed in #452 !

jnm2, 2015-07-06:
I see you are taking the dynamic allocation route after all. Good stuff!
I'm still waiting for the Photons we ordered so I can resolve the problem with the packet boundaries, but it looks like everything is taken care of.

m-mcgowan, 2015-08-22:
The fix has been merged into develop for anyone that's available to test.

m-mcgowan, 2015-08-30:
Tested working on the NTP client in `user/tests/wiring/networking`

closed 2015-08-30 by m-mcgowan