particle-iot/device-os #409: Building spark-firmware for PLATFORM_ID=4

javier-rodas, 2015-04-06:
Hello,

I'm trying to build the bootloader and firmware for the *PLATFORM_ID=4*, that is, the "Spark core-v2/WICED" platform that will be needed for the Spark Photon, P0 or P1 modules.

With today's repository (April 6, 2015) files from the *featured/hal* branch it cannot be built due to the following errors:

1. The ```build/arm/startup/startup_stm32f2xx.s``` file should be renamed to ```build/arm/startup/startup_stm32f2xx.S``` in order to avoid errors building the bootloader.

2. The ```hal/src/core-v2``` directory is missing and not referenced during the firmware compilation when choosing this *PLATFORM_ID=4 (core-v2)*, only the ```hal/src/core-v1``` which includes the header file ```"stm32f10x.h"``` instead of the ```"stm32f2xx.h"``` needed for that specific *PLATFORM_ID*. Renaming the references inside ```hal/src/core-v1``` files doesn't help and returns in the following error:

```
arm-none-eabi-g++ -DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_ID=4 -DSPARK_PLATFORM -DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_ID=4 -DINCLUDE_PLATFORM=1 -DSPARK_PLATFORM_NET=WICED -DPRODUCT_ID=65535 -DPRODUCT_FIRMWARE_VERSION=65535 -DUSE_STDPERIPH_DRIVER -DDFU_BUILD_ENABLE -DRELEASE_BUILD -I./inc -I./src -I./libraries -I../wiring/inc -I../services/inc -I../communication/lib/tropicssl/include -I../communication/src -I../hal/inc -I../hal/shared -I../hal/src/core-v2 -I../hal/src/core-v1 -I../platform/MCU/STM32F2xx/CMSIS/Include -I../platform/MCU/STM32F2xx/CMSIS/Device/ST/Include -I../platform/MCU/STM32F2xx/SPARK_Firmware_Driver/inc -I../platform/MCU/STM32F2xx/STM32_StdPeriph_Driver/inc -I../platform/MCU/STM32F2xx/STM32_USB_Device_Driver/inc -I../platform/MCU/STM32F2xx/STM32_USB_Host_Driver/inc -I../platform/MCU/STM32F2xx/STM32_USB_OTG_Driver/inc -I../bootloader/src/core-v2 -I. -MD -MP -MF ../build/target/main/platform-4/main.elf.d -ffunction-sections -Wall -Werror -Wno-switch -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -g3 -gdwarf-2 -Os -mcpu=cortex-m3 -mthumb   ../build/target/main/platform-4/src/application.o ../build/target/main/platform-4/startup/startup_stm32f2xx.o ../build/target/main/platform-4/startup/startup_stm32f2xx.o --output ../build/target/main/platform-4/main.elf -T../build/arm/linker/linker_stm32f2xx_dfu.ld --specs=nano.specs -lc -lnosys -u _printf_float -Wl,-Map,../build/target/main/platform-4/main.map -T../build/arm/linker/linker_stm32f2xx_dfu.ld --specs=nano.specs -lc -lnosys -u _printf_float -Wl,-Map,../build/target/main/platform-4/main.map -L../build/target/wiring/arm/ -L../build/target/services/arm/ -L../build/target/communication/platform-4/ -L../build/target/hal/platform-4/ -L../build/target/platform/platform-4/ -Wl,--start-group -lwiring -lservices -lcommunication -lhal -lplatform -Wl,--end-group -nostartfiles -Xlinker --gc-sections
```
NOTE: Here there are two includes, and the second is wrong: ```-I../hal/src/core-v2 -I../hal/src/core-v1```

```
arm-none-eabi-g++: fatal error: /usr/local/gcc_arm/gcc-arm-none-eabi-4_9-2014q4/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/lib/nano.specs: attempt to rename spec 'link' to already defined spec 'nano_link'

compilation terminated.
make[1]: *** [../build/target/main/platform-4/main.elf] Error 1
...
```
```
$ arm-none-eabi-g++ --version
arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 4.9.3 20141119 (release) [ARM/embedded-4_9-branch revision 218278]
```

I'm trying to build it from its root folder with the following command: ```make PLATFORM_ID=4 v=1```

Is there any workaround to avoid this build problem?

Are you planning to fix it and merge the core-firmware feature/hal branch to the master soon?. 

And, any comment about when are you planning to update the "firmware upgrade" documentation showing how to low-level upgrade the bootloader and firmware for the P0 / P1 modules? (or Spark Photon in general).

I bought many Spark Cores and I'm now interested in using this spark-firmware with your P0/P1 modules.

Any help would be appreciated, and thanks in advance for your comments.
You are doing and superb job with your Spark dev platform.

m-mcgowan, 2015-04-06:
Hey @javier-rodas 

Thanks for your encouragement!

The code in the firmware repo for PLATFORM_ID=4 is many months out of date and isn't supported at present. The WICED SDK sources are presently non-redistributable, so we've been developing on the WICED platform in a private repo - we are eagerly waiting for a distributable license so we can move all our work back into the public repo. As soon as that happens you'll be able to build firmware for the Photon/P0/P1 and see all the exciting things we've been working on the past few months!

javier-rodas, 2015-04-06:
Thanks for your quick response @m-mcgowan 

I had registered at the Broadcom webpage long time ago, and tested the WICED-SDK 3.1.2 using a WICED BCM43362 dev kit, which if I'm not wrong includes the same WiFi SoC as the one in the USI module you are using for the Photon/P0/P1.

I also found out that the **BCM9WCDUSI09** *platform* [[LINK]](http://community.broadcom.com/community/wiced-wifi/wiced-wifi-forums/blog/2014/11/04/bcm9wcdusi09-platform-files-for-sdk-3x) can be added to the latest version of the WICED SDK 3.1.2 and that it would, in principle, give support for your Photon/P0/P1 modules, right?.

But my question is how could I compile the Spark-firmware in advance, before you provide your Photon/P0/P1 on May. Is it possible?

Could you provide detailed building and re-flashing instructions for the bootloader and Spark firmware when having the WICED-SDK? I guess you do not have to redistribute any licensed code from Broadcom (WICED-SDK sources) for that.

Thank you in advance for your support.

m-mcgowan, 2015-04-06:
Building from source and compiling your own firmware will require the photon source repo, which we can disclose once we have the license. Until then, I can provide you the binary image that is flashed to all P0/P1 modules if that's useful to you. However, you wont be able to compile your own firmware.

javier-rodas, 2015-04-06:
Thanks for your answer @m-mcgowan 

The Photon binary firmware would be nice for start. How do you think it is the best way to share it online?

And do you think I will take you very long to obtain that Broadcom license to disclosure the Photon firmware source-code?
I ask because I have found an important bug at the Spark-Core firmware when working with a local Spark Cloud server that would be interesting to fix also for the Photon.

kennethlimcp, 2015-04-07:
@dljamo can you help @javier-rodas with this privately?

m-mcgowan, 2015-04-07:
@javier-rodas - could you post details of the bug as a separate issue? There's a lot of common code shared by the core and the photon, so it may be possible to fix it for both even though you don't have the photon files.

javier-rodas, 2015-04-07:
@dljamo @m-mcgowan 
I have posted actually two bugs in a separated issue.

m-mcgowan, 2015-04-10:
Thanks! Closing this since the photon build is very out of date in the public repo.

closed 2015-04-10 by m-mcgowan