particle-iot/device-os #410: Spark-Firmware not handling correctly Spark-Cloud server interruptions

javier-rodas, 2015-04-07:
I have found that the latest version of the spark/firmware (0.3.4+) do not support correctly the following situations. I've tested it with a local Spark-Cloud server but maybe it also applies to your global Spark-Cloud server:

1. A core is already connected correctly with the Cloud (breathing cyan) => We stop/kill the Spark-Cloud => The core disconnects and remain in an inconsistent state forever. Even when we restart the Spark-Cloud the core is not able to reconnect again, and it fails due to 30s timeouts continuously. Only a manual reset (**pressing the Reset button or by a power-cycle**) would restore it to a consistent state and it would then connect again correctly.

2. With the Spark-Cloud stopped, we power-up the core and wait a little => The core remains in the same inconsistent state forever (when not founding a server), and no matter if we start the Spark-Cloud afterwards, the core is not able to connect to it and will retry forever after 30s timeouts. Again, only a reset or power-cycle would restore the core to a consistent state.

I have implemented the following **workaround** to force a soft-reset of the cores by firmware, when it detects that it goes through these states, to handle correctly a possible service interruption by a Spark-Cloud.

Class ```main.cpp``` of the ```core-firmware```:

```
...
+ uint8_t flag = 0;

...
#if defined (RGB_NOTIFICATIONS_CONNECTING_ONLY)
		LED_Off(LED_RGB);
#else
		LED_SetRGBColor(RGB_COLOR_CYAN);
		LED_On(LED_RGB);
		SPARK_LED_FADE = 1;
+			//////////////////////
+			flag = 1;
+			//////////////////////
#endif
	}
	else
	{
		LED_Toggle(LED_RGB);
		if(SPARK_CLOUD_SOCKETED) {
			TimingLED = 50;	        //50ms
+			//////////////////////
+			flag = 1;
+			//////////////////////
		} else {
			TimingLED = 100;        //100ms
+			//////////////////////
+			if (flag==1) {
+				NVIC_SystemReset();
+			}
+			//////////////////////
		}
	}
```

I repeat that this is a workaround, so that this behaviour should be fixed without forcing a soft-reset.

I guess all Spark-Cores and Spark-Photon/P0/P1 firmwares are affected by these two issues.

javier-rodas, 2015-04-14:
@m-mcgowan Do you have any comments about these two issues I posted 7 days ago?

Does this behaviour only appear in your "public repository" local Spark-Server implementation (https://github.com/spark/spark-server/) or it also affects to your global Spark-Server?

Do you have any plans to improve the core and/or photon firmwares to correctly support these 2 situations by the "public repository" Spark-Server?... or improve the "public repository" Spark-Server code to correctly deal with them?

Any advice and comment will be really appreciated.

m-mcgowan, 2015-04-14:
Thank you for this. I see this issue, but what was the other issue you posted? I don't recall seeing that.

javier-rodas, 2015-04-14:
Sorry, I meant two issues in the same post, this one.

m-mcgowan, 2015-04-14:
Hi @javier-rodas  

> Does this behaviour only appear in your "public repository" local Spark-Server implementation (https://github.com/spark/spark-server/) or it also affects to your global Spark-Server?

Given that the cloud will be disconnected from time to time (e.g. network outages) and we've had no reports of cores not coming back online seems to indicate that this is an issue only with the local cloud. 

> Do you have any plans to improve the core and/or photon firmwares to correctly support these 2 situations by the "public repository" Spark-Server?... or improve the "public repository" Spark-Server code to correctly deal with them?

Once we are done with all our highest priority issues, we will certainly look into this issue to find a fix for the behavior you are seeing.

javier-rodas, 2015-04-14:
Hi again @m-mcgowan 

Do you think I should open a copy of this issue in the Spark-Server github repository?

emilyrose, 2015-04-14:
@javier-rodas hey! I'm the maintainer of spark-server. 

If you'd be willing to provide steps to reproduce, I would love for you to submit an issue!
I can confirm that I've not seen this behavior on spark-server, but I am happy to attempt to reproduce this with your help.

Thanks!

javier-rodas, 2015-04-14:
Hi @emilyrose  Thank you for replying so quickly.

The steps to reproduce the issues are detailed in the first post of this github issue. I detail the first issue in little more detail:

The set-up I have used was a laptop with the spark-server running over OS X, connected to a regular WiFi access-point (TP-Link), using a WPA2 configuration.

On the other side I had several Spark Cores with the latest firmware (0.3.4), correctly configured with the WiFi credentials.

At this point everything works fine if I **first** start the Spark-Server and **then** I power-up the Cores. All of them correctly connect to the Spark-Server, they periodically publish data, etc.

To reproduce the first issue, you have just to kill the Spark-Server (started with ```node main.js```) with CTRL-C, and watch the LEDs of the Spark-Cores (which were previously connected and breathing-cyan). They disconnect and the LEDs start flashing in other colors: cyan, red, etc. While the server is stopped, the cores remain in these two disconnected/error states/colors trying forever, which is normal...

What it's not normal is what happens when I start the local Spark-Server again, **without** manual reseting of power-cycling the Cores.

The cores try to reconnect, but they fail continuously, and a timeout error message appears in the Spark-Server log console. Sorry but I do not currently have the Cores here to paste you the error message. I remember something like a wrong core identifier was received (due to a protocol handshake error) and then after 30s an error appeared due to a timeout. And the core remains in this state trying forever, and failing in the server with 30s timeouts.

If I manual reset the cores or power-cycle them, they are able to connect and publish data again.

---

You can also reproduce the second issue posed in the first part of this post, **first** powering-up the cores and **then** (after a while) starting the Spark-Server.

mtnscott, 2015-04-14:
@javier-rodas Are you guarding your publish and other cloud function calls with tests for an established cloud connection prior?

javier-rodas, 2015-04-14:
Why should I test if the cloud connection before publishing data?

I don't think a call to publish() without a connection established with the cloud can hang the core, or deal with any other problem. These publish calls would be simply missed, am I wrong?.

And, as I said, in the server log appears a wrong core identifier handshake error (or similar), in the two situations I've posted.

mtnscott, 2015-04-14:
It's best practice if you know a function call is going to timeout to not call that function.  The Spark Core is very sensitive to timing and I'm not sure that calling functions from within your loop that will timeout won't cause problems elsewhere.

I keep flags that represent the current WiFi and Cloud state.  If I have lost my Cloud connection I typically don't call functions that rely on a cloud connection and I have functions that will attempt to re-connect to the cloud it the connections are lost.  With that said I manage my own network connections with SYSTEMODE(MANUAL).  

You mentioned you were getting flashing RED, that typically happens when the system goes into a panic state - maybe caused by running out of RAM.  It could be that you are causing the core to queue up messages and it just runs out of free RAM?

I have not looked at the publish firmware code.  I'm not sure if it is making a copy of your message, looking at the message length, keeping a handle to your memory, etc.  I do think the function returns and later attempts to send the message, so calling it again before it has sent the previous message could be bad.

Just some thoughts...

javier-rodas, 2015-04-14:
Hi @mtnscott Thanks for your thoughts.

Maybe you are right and the Core enters in panic mode, but it should recover correctly after panic if the firmware and/or Spark-Server are robust. It's what everybody should expect from a IoT platform.

The problem here is that it seems that the Spark-Server doesn't handle correctly the **core identification or the handshake after being restarted (not after a network issue)**, while the core is already running in a "disconnected from the cloud" state.

On the other side there is no place in the documentation page, as far as I know, that recommends you to test for the network and cloud connection before each network call. The ```SYSTEM_MODE(AUTOMATIC);``` should manage all of this by default, I imagine.

mtnscott, 2015-04-14:
@javier-rodas  - I have a suggestion, you could download my DNS tool and manually control your network and cloud connection.  I'm curious if you can recover the cloud connection by turning it off, and turning it on again? - maybe even trying to turn on the WiFi and turning it on again.

Here is a link to the tool - https://github.com/piettetech/DNSTest

I agree that `SYSTEM_MODE(AUTOMATIC)` should manage everything correctly for you.  Not sure why it's not but maybe if we can isolate the problem we have an idea of where to start to look.

javier-rodas, 2015-07-10:
Hello @emilyrose and @kennethlimcp 
I have tested the local Spark-Server with the Photon and the same user-application firmware I was using before with some Cores, this time on top of a develop 0.4.2 firmware => **With the Photon the behaviour commented in this issue doesn't happen**.

For your information, first of all, to had to make the local Spark-Server "compatible" with the Photon applying the override patch that @kennethlimcp commented in the following issue: https://github.com/spark/spark-server/issues/45   that is, commenting the lines 343 to 346 in the following file: https://github.com/spark/spark-protocol/blob/master/js/lib/Handshake.js#L343

After this point, everything worked as expected, as they were working with the Cores. The event notifications and functions calls work as expected with the Photon and the local Spark-Server.

In more detail, regarding the new local Spark-Server interruptions behaviour, now the Photon goes from breath-cyan to blink-cyan state when the server is stopped, **and after not being able to connect to the server the Photon blinks ONE time and very quickly in RED**. After this, it retries after 1s, then after 2s, then after 4s, 8s, and son on.

If the Spark-Server is launched in between these retries, the Photon won't connect until the corresponding timer expires, that's normal. But, the potential problem for me here is that **this retry time doubles after each retry** and if the Spark-Server is stopped for several minutes and started again, **the Photon couldn't connect after SEVERAL minutes**.

My question here is the following. Where can this **retryTime_(t) = 2*retryTime_(t-1)** behaviour in the **spark-firmware develop branch** can be modified=? I want to fix a lower maximum retry-limit of 30s for example, to not let this retry-time to increase up to few minutes?

kennethlimcp, 2015-07-11:
@javier-rodas , that's the behavior coded in the firmware so what you observed is correct not only on a local :cloud: but also the Particle :cloud:. Let's see what the firmware team says.

javier-rodas, 2015-07-12:
@kennethlimcp @m-mcgowan Thanks for the information.

I guess you should review the line 222 in 8195de2, because you use ```cloud_failed_connection_attempts``` instead of the local parameter name  ```connection_attempts```. Anyway the behaviour is correct as you increment the ```cloud_failed_connection_attempts``` variable "globally" and only in one place.

```
unsigned backoff_period(unsigned connection_attempts)
{        
    return 500*((1<<min(9,cloud_failed_connection_attempts))-1);
}
```

should be:

```
unsigned backoff_period(unsigned connection_attempts)
{        
    return 500*((1<<min(9,connection_attempts))-1);
}
```

On the other side there is another deprecated comment on line 216 of the same ```system_task.cpp``` file. When using the previous code using a 500 multiplication factor:

```
 * Series is 0, 100, 300, 700, 1500, 3100, 6300... up to 409500
```

should result in:

```
 * Series is 0, 500, 1500, 3500, 7500, 15500, 31500, 63500, 127500 and 255500.
```

So the maximum ```backoff_period``` after 9 attempts is currently fixed to **255.5 seconds (4 minutes and 15 seconds)**.

---

For me, the right way to write this exponential sequence would be the following, and I would stop after 7 attempts, as 32 seconds is in my opinion a reasonable maximum retry interval to deal with typical Internet / LAN network issues:

```
 * Series is 0, 500, 1000, 2000, 4000, 8000, 16000 and 32000.
```

```
unsigned backoff_period(unsigned connection_attempts)
{        
    return 500*(1<<min(7,connection_attempts-1));
}
```

Finally, this code could be simplified even more, as you increase at line 211 the ```cloud_failed_connection_attempts``` by 1 each time a ```cloud_connection_failed()``` is launched when ```cloud_failed_connection_attempts<255```. You could just compare it with a constan number of 9 (or 7 in my case), and you wouldn't need to perform the ```min``` call at line 222, saving some CPU clock cycles :neckbeard: 

The code could be like this:

```
/**
 * The number of connection attempts.
 */
static uint8_t cloud_failed_connection_attempts = 0;

void cloud_connection_failed()
{
    if (cloud_failed_connection_attempts<7)
        cloud_failed_connection_attempts++;    
    cloud_backoff_start = HAL_Timer_Get_Milli_Seconds();
}

/**
 * Series is 0, 500, 1000, 2000, 4000, 8000, 16000 and 32000.
 * @param connection_attempts
 * @return 
 */
unsigned backoff_period(unsigned connection_attempts)
{        
    return 500*(1<<connection_attempts-1);
}
```

m-mcgowan, 2015-07-14:
Thank you for this.  The mix-up of using the global value rather than the local value is an oversight and I'll fix this, although in practice the two are always the same, so there is no consequence in practice. The intent is that `backoff_period()` is self-contained so we can unit test it independently from the state of other global variables.

Artificially limiting `cloud_failed_connection_attempts` to 7 to avoid the `min` comparison is premature optimization, which I doubt would make any measurable difference (you're welcome to measure the difference if you feel it would!) 

Assuming it is even a few instructions faster, the bigger issue is that it impinges on the design and expressivity of the system. In this case, we no longer have an actual counter of connection attempts, since it's now artificially limited to 7 to prevent the backoff period from increasing.  For example, if we wanted to introduce logic that after 10 failed connection attempts we re-initialize WiFi that would be problematic with the limit on the failed connection count.  The code to compute the backoff period is no longer maintained in one function but decoupled and spread out over several, which to my mind is not an improvement. When optimizing, I find it helps to keep in mind the difference between Fast and Fast Enough! ;-)

javier-rodas, 2015-07-14:
:+1:  You are completely right. I didn't realised that you compute the backoff period in several functions, so in that case it makes not sense to limit the ```cloud_failed_connection_attempts``` to 7 in the ```cloud_connection_failed()``` function.

I guess you can close this issue now.

m-mcgowan, 2015-08-23:
I've made the backoff period less agressive - it will start with 0.5s, 5 times, then 1s, 5 times, then 2s, five times, doubling the timeout after five tries. This is in the 0.4.4 firmware. 

Going to close this as suggested. Thanks!

closed 2015-08-23 by m-mcgowan