particle-iot/device-os #411: Spark and Arduino boards manager?

jacobrosenthal, 2015-04-13:
Is the board manager, putting spark in the arduino ide, something Spark will ever be interested in?
http://blog.arduino.cc/2015/04/09/install-intel-galileo-edison-with-the-ide-boards-manager/

zsup, 2015-04-13:
I expect it's not possible to do so because we use a totally different toolchain, but if it is possible, then definitely interesting!

jacobrosenthal, 2015-04-13:
Seems to me it supports alternate compilers and uploaders.

plocher, 2015-08-06:
Is the compiler toolchain available "locally" - or is it only a web service?
Either way, it should be doable - and not too difficult.  

The "gotcha" will be that the Photon/Spark environment isn't really Arduino compatible, so it may well be that this is a "tip of the iceberg" suggestion - blink works, but anything with a complex library probably won't, sketches with .cpp/.h files won't compile  (Application.h -vs- Arduino.h...), etc.

IMO, a useful first step would be coming up with a roadmap that sets some scope, context and boundaries:  should Spark be Arduino compatible? what gaps and incompatibilities stand in the way of Spark becoming more Arduino compatible?, how much "compatible" is good enuf?  is there such a thing as being "too compatible"?, how/where do the Spark extensions fit into the picture?  How should libraries be ported/managed?...  yadda...

zsup, 2015-08-07:
The compiler toolchain is available locally, see www.github.com/spark/firmware (the `develop` branch has all the latest stuff, much of which will soon be merged into `master`)

Generally speaking I think making Particle and Arduino more compatible would be awesome. When we first started building our platform, Arduino didn't have a library manager or a board manager, so we had to diverge from Arduino's path to do what we wanted to do (use a non-Atmel processor and provide an easy path for importing libraries). These days the state of the world is quite different, but we've already built a bunch of stuff and made different architectural decisions. If someone was interested in helping to bring our paths closer together, I would be very interested in seeing it happen, although I worry at this point it might be a pretty massive project.

rei-vilo, 2015-08-15:
@zsup 

Don't you remember the tickets #194 #195 #196 #197 #198 #199 #200 #201 #202 I created for a better Arduino compatibility?

The `develop branch` for the Photon is very different from the `master branch` for the Core, even from the `feature/hal` I'm using with [embedXcode](www.embedXcode.weebly.com) on Xcode.

aleksandrguru, 2016-01-13:
when you can work with Arduino Ide?

rei-vilo, 2016-01-13:
I've given up. I'm no longer using Spark / Particle boards except for basic IoT projects.

aleksandrguru, 2016-01-13:
sorry, I bought PHOTON only because he thought that would be compatible with the Arduino, it is good that now there ESP8266.Chastitsa stall with time

dmiddlecamp, 2016-01-13:
Just wanted to chime in, I'm sorry you've given up @rei-vilo .  We haven't forgotten about this, and it's still important to us as a team.  Most Arduino libraries and examples compile easily or require minimal changes in the IDE or Particle Dev, but I'd love to see full compatibility in the Arduino IDE.  I'm sorry it hasn't happened yet, but we haven't given up.  :)

aleksandrguru, 2016-01-13:
I can not tolerate this web ide))I hope everything will turn out

rei-vilo, 2016-01-13:
Thanks for the update, @dmiddlecamp.

I'll resume using Spark / Particle for more complex IoT projects once I can use a the tool-chain with a standard IDE and build and link on command-line mode.

Good luck with the development!

plocher, 2016-01-13:
David,

It's an attitude thing, y'know?  Arduino compatible means much more than
"developers can port their code to a different environment where sketches
won't compile without changes"...

Everyone else in the ecosystem is fully on board with the idea of
leveraging the Arduino IDE, tool chains, board manager and the like, yet
the Particle team has chosen to fork the environment and go it alone,
thinking there is value in reinventing the infrastructure and forcing
developers to migrate their workflows and change their code.  Braap.

Marketing 101 teaches us that incompatible innovation must be 100x better
than the status quo for people to be willing to replace the status quo.
Two orders of magnitude...

Right now Photon has a slight edge up on things because of the built in
wifi and easy IoT.  Unfortunately, the negative baggage of being different
mostly offsets that advantage, especially with the ESP8266 devices being an
"almost" replacement.  When the ESP32 comes out, fully supported by the
'duino ecosystem, things will just get worse.

If you really want Particle to succeed (we do!) how about upping this
priority from a passive desire to a resourced and scheduled scrum
deliverable?

  -John

dmiddlecamp, 2016-01-13:
Hi @plocher,

  Thanks for the thoughtful reply!  I don't think we've forked the Arduino IDE or anything, I think instead like you said we've tried to take something very different, and make it accessible to as many people as possible.

I agree that we should prioritize this, a big priority this quarter is us focusing on the stability of our tools and developer support.  I'll bring this up with the team and see if we can get this moving again.  Thanks for being a passionate supporter, it's very much appreciated :) 

Thanks!
David

rei-vilo, 2016-01-13:
@plocher 
Give the MediaTek LinkIt One a try. It comes with WiFi, Bluetooth 2.0 + BLE, GPRS + GMS, GPS and full Arduino IDE support. 
I've reviewed it on my website at [MediaTek LinkIt One](http://embeddedcomputing.weebly.com/mediatek-linkit-one.html).

clay-to-n, 2016-01-14:
@rel-vilo With regards to your desire for an IDE and command line tools - have you tried using [particle-cli](https://github.com/spark/particle-cli)?  I code in Sublime Text (but you could use any other editor or IDE) and use the cli tool for flashing code to my Photon.  A much better experience than the Arduino IDE in my opinion.

plocher, 2016-01-14:
The difference between using an ESP8622 and a Photon in an "Arduino
ecosystem" is the difference between "compatible, with improvements" and
"incompatible and different, too".

With the ESP/Huzzah/..., or one of the various SAM ARM boards or ..., I can
choose to use the Arduino IDE, add in the required board manager plugin,
and things "just work".  Period.
If I hate the Arduino IDE, I can use the Arduino CLI in a makefile - which
handles code and libraries identically to the IDE, but lets me use my own
IDE - Eclipse, Sublime or (gasp) vi :-)
Or I can use the various Web IDEs (albeit with some rough edges...)
By providing a board "plugin", users are free to choose how they interact
with your product.

With Photon, I can't do any of that.  The Arduino IDE, CLI and Web tools
could support Photon, but Particle doesn't provide a board definition and
compiler toolchain that I could plug in.  Users are forced to use a
different system to develop code for the boards.

Bastardizing a C/C++ idiom,  the first example is "Arduino++", while the
second is "--Arduino++".

  -John

rei-vilo, 2016-01-14:
@clay-to-n 

Yes, I've already gave a try but faced some issues with the makefiles, especially during linking for Photon. 

As a matter of facts, everything runs fine for the Core based on the `feature-hal` branch. Unfortunately, it doesn't support the Photon. The new recommended branch `Latest` breaks everything. 

See [Local building for Core+Photon](https://community.particle.io/t/local-building-for-core-photon/11221) and [Particle on Xcode with embedXcode](https://community.particle.io/t/particle-on-xcode-with-embedxcode/8928) on the forum.

rei-vilo, 2016-02-04:
Please have a look at what **RedBear** has achieved for the Duo board, incidentally relying on Particle for the IoT infrastructure, at [STM32-Arduino](https://github.com/redbear/STM32-Arduino). 

RedBear can claim true Arduino-compatibility.

dmiddlecamp, 2016-02-04:
Hi @rei-vilo,

  Thanks for the bump!  Yes we're aware of their efforts, sorry about the delay on this, and it's certainly not incidental, we've been working closely with RedBear Labs.  :)

Thanks,
David

rei-vilo, 2016-02-04:
@dmiddlecamp You're welcome!

I can imagine RedBear and Particle worked on close cooperation. Good luck with the implementation for the Particle boards!

daved, 2016-03-06:
Will this change also cover the Electron?