particle-iot/device-os #414: Start up condition of I/O pins

kennethlimcp, 2015-04-18:
There is a long conversation in the community with regards to clearly know the pins behavior during startup: https://community.spark.io/t/start-up-condition-of-i-o-pins/10556

We will need to state down clearly in documentation of how this is being defined in code. 

Questions to be addressed
- Behavior of pins upon startup and before user code executes
- Behavior of pins upon reset
- Behavior of pins during a cloud disconnection in AUTOMATIC mode (since user code might not be running)

Ping firmware team: @m-mcgowan @towynlin @pkourany @satishgn @pkourany

kennethlimcp, 2015-04-19:
From http://www.st.com/web/en/resource/technical/document/reference_manual/CD00171190.pdf

```
The JTAG pins are in input PU/PD after reset:
PA15: JTDI in PU
PA14: JTCK in PD
PA13: JTMS in PU
PB4: NJTRST in PU
```

m-mcgowan, 2015-04-19:
to the best of my knowledge we don't change much regarding the pin states, and use whatever the default state is for the stm32. The one exception applies to the jtag pins which are remapped early on during start up to regular digital pins. https://github.com/spark/firmware/blob/master/src/main.cpp#L93

technobly, 2015-04-20:
Just to elaborate, the function that Mat points out fully disables the JTAG-DP and SW-DP debugging interface, but it only does so if `SWD_JTAG_DISABLE` is defined.  `SWD_JTAG_DISABLE` is defined by default when `USE_SWD_JTAG` is not defined (also the default state).  `USE_SWD_JTAG=y` would need to be added to the command line to not call this function, which would keep these pull-up/pull-down resistors active for JTAG/SWD debugging.

That said, this SWD disable function might be called sooner to shorten the duration of the pull-up/pull-down resistors.

Another way to counteract these pull-ups/pull-downs is to design your circuitry to require more current than the approx 40k ohm pull-up/pull-down resistor provides.  Thereby making your circuit unresponsive to a weak pull-up/pull-down.

If the above is not possible, it would be advised to add a low value pull-up/pull-down in the opposite direction than the JTAG interface is enabled by default.  E.g., if there is a 40k ohm internal pull-up on JTDI (D5) by default that affects your circuit, add a 1k ohm pull-down to completely counteract this 40k ohm pull up.  The following table lists all possible combinations:

Core/Photon Pin | Description | STM32 Pin | Default internal<sup>[1]</sup> | To counter affect
:---:|:---:|:---:|:---:|:---:
D7 | JTAG_TMS | PA13 | 40k pull-up | add 1k pull-down
D6 | JTAG_TCK | PA14 | 40k pull-down | add 1k pull-up
D5 | JTAG_TDI | PA15 | 40k pull-up | add 1k pull-down
D3 | JTAG_TRST | PB4 | 40k pull-up | add 1k pull-down

**Notes:**
<sup>[1]</sup> Default state after reset for a short period of time before these pins are restored to GPIO (if JTAG debugging is not required, i.e. USE_SWD_JTAG=y is not specified on the command line.

pkourany, 2015-04-20:
@technobly, great write-up.  It is important to note that there is a cost to using 1k pull-up or pull-downs in the increased current draw (3.3ma) on the pins where the resistors are connected and being driven in a state opposing the resistor.

technobly, 2015-04-20:
Great point @pkourany!  I was thinking of that as well, but thought that a 1k is most definitely going to ensure a LOW output on a pin with a 40k ohm pull-up.  Calculating some things, a 4.7k ohm pull down should still represent a low 3.3V TTL output with a 40k ohm pull-up, but this is also highly dependent on what the external circuitry is.  I updated the chart above to say ≤ 4.7k.

steelydev, 2015-07-28:
If possible, I would vote the firmware default AS EARLY AS POSSIBLE to OUTPUT, LOW on all I/O pins, which seems safe AND an easy design target.

bkobkobko, 2015-07-29:
Hi @steelydev 

Generally I/O pins should be set to INPUT at reset.  The JTAG pins are special in that if you ever want to use JTAG debugging they need to be set a certain way for a brief time at power-on.

andyw-lala, 2015-07-29:
The pins should be input (high impedance) until the firmware configures
them correctly.

It should be the responsibility of the external circuitry to pull lines
high or low if required, for example by a resistor.

steelydev, 2015-07-29:
@bkobkobko As usual, you seem to be correct.  I retract my vote.  Thank you for your direction!

@andyw-lala I modified my prototype PCB with a bunch of 4.7K resistors all pulling to GND and it has solved unwanted behavior.

m-mcgowan, 2015-08-22:
I'm not sure how we can act on this, since the advice seems to be leave the pins as high-impedence INPUT state (which they are set to as soon as we can execute code) and that the external circuitry should pull lines to the desired state.  Closing this now, please re-open if there are further points to discuss.

closed 2015-08-22 by m-mcgowan