particle-iot/device-os #415: functions to wait for a given state

m-mcgowan, 2015-04-19:
Users often want to wait for something to happen - wifi to connect, the cloud to connect etc..

At present, the way to make this happen is to have a loop that checks the state and runs background processing:

```cpp
// wait for wifi to be ready

while (!WiFi.ready()) {
   Spark.process();   // replaces SPARK_WLAN_Loop in firmware v0.4.0
}
```

Code like this is fragile since it doesn't handle the case where wifi doesn't connect.

It would be more convenient for users to be able to issue a one-liner to wait for the state or timeout:

```cpp
if (WiFi.waitFor(READY, 10000)) {
   // wifi ready
}
else {
   // timeout - wifi not ready
}
```

The code is simpler, clearer, and the system can take care of background processing as needed, plus the system can initiate any tasks needed to get the system in the desired state. In this example, if WiFi were not connecting/connected, then it could initiate `WiFi.connect()` to begin a wifi connection.

m-mcgowan, 2015-06-05:
With Lambdas, we could make this generic:

```
waitFor([]->{ WiFi.ready(); });
```

Is the syntax a bit too spooky? We could wrap it up in a macro:

```
WAIT_FOR(WiFi.ready());
```

pkourany, 2015-07-03:
I like the lambda approach.  It also ends up being an inline and "disposable" function I believe :)

m-mcgowan, 2015-08-23:
With multithreading, waiting for an event is a common need. Using `waitFor`, we will be able to avoid busy-waiting and use condition variables instead in the long term.

m-mcgowan, 2015-08-23:
Here's the proposed syntax:

```cpp

 // waits for WiFi.ready() to be true for up to 10 s
wait_for(WiFi.ready, 10000);  

// wait and check if WiFi was ready 
if (wait_for(WiFi.ready, 10000)) {
   // wifi is ready
} else {
   // wifi is not ready
}

// no timeout
wait_until(WiFi.ready);

// all the above are syntactic sugar for
// what advanced C++ coders might prefer 
System.waitCondition([]{ return WiFi.ready(); }, 10000);
```

zsup, 2015-08-23:
Should `wait_for` and `wait_until` be `waitFor` and `waitUntil`? Given that the rest of the Arduino syntax is camelCase not caterpillar_case, I'd expect these to be camelCase.

I'd be really curious to see what other methods you're thinking of creating for managing timing, threads, concurrency, etc. in the multithreaded code? Curious how this fits in with the broader set of commands that will be available once multithreading is possible.

m-mcgowan, 2015-08-23:
I agree. I originally had `waitFor` etc.. but thought it was easier to read with the hyphen...all the while having alarm bells ringing about not being consistent with the existing API conventions of camelCase. Just wanted to see if this would fly! Happy to use camel case.

m-mcgowan, 2015-08-25:
```
waitFor(WiFi.ready);
waitUntil(WiFi.ready, 10000);
```

andyw-lala, 2015-08-25:
I'm a little puzzled - seems like a bunch of namespace pollution for an
optional timeout argument.

If the timeout parameter is not present, wait forever; if it is present,
obey the timeout.

m-mcgowan, 2015-08-25:
Andy, they're macros, so have to have a different name if we want different numbers of parameters, since macros cannot be overloaded.

andyw-lala, 2015-08-25:
I guess that wasn't clear to me.

I suppose they _have_ to be macros because you're making the wait generic
and dropping in different conditions/lamdas ?

Sure there's absolutely no way to implement using something that detects
presence/absence of timeout ?

m-mcgowan, 2015-08-26:
The main issue is that we are developing a DSL.
```
waitUntil(WiFi.ready);
```
Writing `WiFi.ready` is not a legal syntax in C++. A member function reference would normally require using `std::bind` and `std::mem_fn`, so the syntax quickly becomes bloated. So I chose to use a macro and a lambda to make the syntax much simpler. 

Other alternatives rejected were `waitFor(WiFi, WiFiClass::ready)` which could be implemented as a inline template, and hence overloadable. But it's still not elegant. 

IMHO the simplicity of the proposed waitFor/waitUntil syntax compared to non-macro alternatives outweighs the small downside of having two separately named macros.

andyw-lala, 2015-08-26:
Sure. Seems like it has to be.

closed 2015-08-26 by m-mcgowan

bkobkobko, 2015-08-26:
If you want to have a general purpose waitFor/waitUntil then this almost certainly the way to go, but another approach to solving the same problem would be to add a method to WiFi like

```cpp
   WiFi.waitForReady();
or
    WiFi.waitForReady(1000);
```

If there are only a few things that people want to wait for with timeouts, then this feels better.

m-mcgowan, 2015-08-26:
Sure, I'd considered that too, and haven't ruled it out, but we would then have to capture all state methods:

```
WiFi.waitForConnected
WiFi.waitForDisconnected
WiFi.waitForListening
WiFi.waitForReady

Particle.waitForConnected
Particle.waitForDisconnected

// AP mode
AP.waitForConnected
AP.waitForDisconnected

// and again with the Electron,
Mobile.waitForConnected
Mobile.waitForDisconnected
```

I wanted avoid the redundancy of having a wait state method for each state.

We will need methods to wait for all states with multithreading since some operations that we now assume are synchronous (e.g. WiFi.disconnect()) will become asynchronous.