particle-iot/device-os #416: TCP flush() does not work as expected

pkourany, 2015-04-20:
The existing tcp flush() function only resets the Core buffer pointers but does not flush the CC3000 queue of remaining tcp request return bytes.  Doing a client.stop() does not flush the data from the CC3000 either.  Both conditions lead to the CC3000 buffers eventually filling and causing failure.  The work around in user code is to do read() until no more bytes are available (example only):

while (client.available())  //flush all CC3000 queue bytes prior to closing connection
  char c = client.read();

if (!client.connected())  //if disconnected, close the connection
  client.stop()

m-mcgowan, 2015-04-20:
I'm puzzled about flush(), and hope someone can shed some light. TCPClient inherits from Stream, so flush should behave as you would expect for streams.

The arduino docs list `flush()` as the buffer when all data is sent. - http://www.arduino.cc/en/Reference/StreamFlush.  The description is not entirely clear - which buffer, read data or write data? 

The source code of [EthernetClient.cpp](https://github.com/ninjablocks/arduino/blob/master/Ethernet/EthernetClient.cpp#L122-L125) has this implementation for flush():
```
void EthernetClient::flush() {
  while (available())
    read();
}
```

This seems consistent with our current behavior where the read side of the stream is drained, and implementing it like this would ensure that the underlying socket is also drained.

pkourany, 2015-04-20:
I would say at the stream level, just resetting the queue pointers is fine since Stream does not know the device it is associated with.  For the outgoing queue, the reset pointers will stop further data from being consumed by the device driver.  However for the incoming queue, the Stream flush does nothing to ensure the device's queue has been emptied.  This explains why Arduino's EthernetClient version of flush() actually flushes the device's receive buffer using read().  I believe we need the same thing for the TCP client flush().

m-mcgowan, 2015-04-20:
:+1:  Agreed (we are saying the same thing!)

bkobkobko, 2015-04-20:
I think it is better to wait for data to begin returning and then flush since if you call this directly after say a GET request, the flush ends up thinking it is done when really it has not started yet.

Obviously waiting for return data to flush needs a timeout.  The second loop can have a timeout too but might not need one.

```cpp
  unsigned long startTime = millis();
  while (client.available() == false && millis() - startTime < 1000) {
    // Just wait up to 1000 millis
    }
  startTime = millis();
  while (client.available() > 0 && millis() - startTime < 1000) {
    client.read();
    }
```

m-mcgowan, 2015-04-20:
Thanks bko, I can see the logic behind this, although it's a distinct departure from the Arduino implementation. But gives food for thought!

We should be careful with arbitrary timeouts - they'll work sometimes and not other times, which can lead to devices suddenly behaving differently under network load. At the very least the user should be able to specify the timeout to first wait for data, since it's specific to the situation. 

It's really not clear to me why we need flush() as defined by Arduino. In other languages, flush means wait for the data to be written, which is almost a complete opposite to the Arduino implementation that clears the read buffer.  The problem is the inherent timing issues - available() may intermittently return 0 as data is being streamed, even when some of the data has yet to be sent, causing the flush() loop to exit early.  To me it seems like a flawed design in the Arduino API that's loaded with race conditions. I feel there are better ways to ignore data received than relying upon `available()`. 

E.g. the case with HTTP, rather than skipping bytes until available returns 0, it would be better to properly parse the HTTP response, which encodes the response body length so that the entire response can be fully read and discarded, even when there are delays delivering the data to the device.

So, do we support a broken API for the sake of Arduino compatibility? (NB: presently flush is undocumented.)

bkobkobko, 2015-04-20:
I would not have designed it this way either, but there is a lot of code out there in Arduino land, particularly for the WizNet Ethernet board that uses client.flush().  On the WizNet chip I believe you only have one transmit and one receive packet buffer so flush just clears the hardware buffer.

This is especially for the case where the user **does not** want to parse the HTTP response and see what happened either because they have no room to do so or it is not really relevant since they are going to send new data again in N seconds anyway. An HTTP library should read the response and possibly parse it but I don't think TCPClient should.

For HTTP since you are getting a packet at time and reading only a byte at time, I have never seen available go low in the middle of a response.  I know it could, but it never has for me.

Updating the Spark example for TCPClient to have more realistic data handling would be the first step here, even if you don't want to change the firmware.

m-mcgowan, 2015-04-21:
For now, I've implemented flush() as a read while available. This clears the local TCP Buffer and any data available in the cc3000. At least that gives us arduino compatibility, which is our baseline.

m-mcgowan, 2015-04-25:
Going to close this since the original issue is addressed. We support the same implementation as arduino, and thus the same flaws. We can open additional issues if this later needs refinement, but I feel this is a poorly designed function and should ideally be avoided.

closed 2015-04-25 by m-mcgowan