particle-iot/device-os #417: Instructions for compiling latest firmware on Mac OSX

1sc0, 2015-04-21:
The latest firmware requires GCC 4.8 so instead of:

    brew install gcc-arm-none-eabi

Do:

    brew install gcc-arm-none-eabi-48

If you have already installed it, when you try to do a `make` you'll see errors like:

    ../CC3000_Host_Driver/evnt_handler.c:61:23: fatal error: stdatomic.h: No such file or directory

    ...

    obj/startup/startup_stm32f10x_md.o: In function `LoopFillZerobss':
    core-firmware/build/../startup/startup_stm32f10x_md.S:98: undefined reference to `SystemInit'

    ...

    obj/src/wifi_credentials_reader.o: In function `WiFiCredentialsReader::print(char const*)':
    core-firmware/build/../src/wifi_credentials_reader.cpp:106: undefined reference to `Delay'

In such case first uninstall the old version:

    brew uninstall gcc-arm-none-eabi

Install the correct version:

    brew install gcc-arm-none-eabi-48

And throw in a: 

    make clean all

m-mcgowan, 2015-04-21:
:+1: Thanks for this. Do you know about our OS X make setup notes, linked from the readme.md in the repo - https://gist.github.com/joegoggins/7763637

1sc0, 2015-04-21:
Now I do :)  Thanks!

closed 2015-04-21 by m-mcgowan