particle-iot/device-os #420: The order of subscribe commands makes a difference with web hooks

bkobkobko, 2015-05-07:
I don't know if this is a bug in the firmware or in the cloud (Hi @dmiddlecamp ) but I change the order of two subscribe commands in setup, it either works or does not work.

Both subscribes return true and everything seems fine, it is just that the IP one never fires unless it comes before the other subscribe.

See this thread in the forum for more:  https://community.spark.io/t/automatic-time-zone/11027

Failing case:
```cpp
int lastTime = 0;
char externalIP[150] = "EmptyIP";

void ipHandler(const char *name, const char *data)
{
	strcpy(externalIP, data);
}

void webHookHandler(const char *name, const char *data)
{
}

void setup()
{
	Spark.variable("externalIP", externalIP, STRING);
	Spark.subscribe("hook-response/get_weather", webHookHandler, MY_DEVICES);
	Spark.subscribe("spark/device/ip", ipHandler);
}

void loop()
{
	if (Time.now() - lastTime > 15)
	{
		Spark.publish("spark/device/ip");
		lastTime = Time.now();
	}
}
```

Working case:

```cpp
int lastTime = 0;
char externalIP[150] = "EmptyIP";

void ipHandler(const char *name, const char *data)
{
	strcpy(externalIP, data);
}

void webHookHandler(const char *name, const char *data)
{
}

void setup()
{
	Spark.variable("externalIP", externalIP, STRING);
	Spark.subscribe("spark/device/ip", ipHandler);  //me first!!!
	Spark.subscribe("hook-response/get_weather", webHookHandler, MY_DEVICES);

}

void loop()
{
	if (Time.now() - lastTime > 15)
	{
		Spark.publish("spark/device/ip");
		lastTime = Time.now();
	}
}
```

kennethlimcp, 2015-05-07:
@bkobkobko I can test this out with more verbose turned on for my devices. Will let you know if i get to this

kennethlimcp, 2015-05-31:
@bkobkobko , i tested this out and managed to replicate this behavior.

Also tested out using a Photon and the behavior is consistent. I will dug figure to see what's causing it!

kennethlimcp, 2015-06-02:
@m-mcgowan and @bkobkobko,

i tested this and it seems like for some reason the function handler is not called when the `Spark.function()` is called. I blinked D7 once the function runs and changing the sequence did make a difference.

Long story short, no issues with the :cloud: side since i watched the **trace** and it is also consistent with the photon

markterrill, 2015-06-03:
Hi,

Quick comment that it can be any event name, not just a child node (ie 'something/child'). I can see the hook responses coming through fine on the CLI:
```
{"name":"hook-response/pushtemps/0","data":"{\"name\":\"-JqsRE15YRWVQcZUJJw0\"}","ttl":"60","published_at":"2015-06-03T08:01:58.943Z","coreid":"undefined"}
{"name":"hook-response/gettargets/0","data":"\"320-84-275\"","ttl":"60","published_at":"2015-06-03T08:01:59.870Z","coreid":"undefined"}
```

However on serial I can see that only one function gets called
```
hook-response/gettargets/0, data: "320-84-275"
```

Strangely enough, I even observed behaviour where within the callback associated with the first subscribed webhook, the eventname of the second webhook was being output with the data of the first one. To fix this I changed the function variable names to be unique, ie `void cbPushtemps(const char *eventPushtemps, const char *dataPushtemps)`. You can replicate this by copying the docs myHandler function and then make another copy and rename it. Cue xfiles music.

m-mcgowan, 2015-06-03:
There is a but on 0.3.4 (the current Core firmware) that causes segmented event names (a/b) to be incorrectly registered. This has been fixed in code but not yet released for the core. It's fixed for the photon. 

@kennethlimcp if you have this setup as a test case, would you mind testing against the latest develop branch on either the Core or Photon to verify this has been fixed. Thank you!

kennethlimcp, 2015-06-03:
@m-mcgowan when was it fixed for the photon, i actually did the test using a photon...

m-mcgowan, 2015-06-03:
It was fixed a long time ago (several months). So maybe it's not the same bug?

kennethlimcp, 2015-06-03:
It looks like some combination of `Spark.subscribe()` is not triggering the function to execute though the API was successful.

kennethlimcp, 2015-06-03:
@m-mcgowan , verified that this issue persists on the Photon as per fbf2b907e1df923a36bba94375356aebfbac436c

m-mcgowan, 2015-07-16:
I just tried the code above (failing case) on the Photon with the current develop branch (https://github.com/spark/firmware/commit/3cf97176e1dc3c034216cda159c262ae6ad54f69) I print the received IP to serial, and am seeing the IP address set. 

There was a problem with `spark/" events that were not being passed to the application handler that was fixed in 0.4.3.

Please test and let me know the outcome. Thanks!

m-mcgowan, 2015-08-22:
Anyone available to test this against 0.4.4?

m-mcgowan, 2015-09-13:
Verified working regardless of the order events are subscribed to.

closed 2015-09-13 by m-mcgowan