particle-iot/device-os #421: Roadmap for migration from SPARK to PARTICLE?

rei-vilo, 2015-05-13:
Is there a roadmap for the migration of the libraries, functions, variables, constants from `SPARK` to `PARTICLE`?

m-mcgowan, 2015-05-31:
All our exisitng APIs will be available with their Spark names for at least 6 months, and any APIs that will be removed will be given notice ahead of time.

We will draw up a roadmap for this in the coming weeks.

towynlin, 2015-07-27:
For all public-facing APIs, we should make `Particle` available as an alternative soon. Beginning with the release of the Electron in October, all documentation should refer to `Particle` not `Spark`. If we can add compile deprecation warnings (not errors) at that time too :+1:. Sometime shortly after that, perhaps the end of the year, we should remove `Spark` entirely.

m-mcgowan, 2015-08-10:
:+1:  Initially `Spark` in firmware spanned a lot of functionality. With 0.4.x, it's been pared down to just cloud connectivity, so wanted to mention the possibility of calling it *Cloud*. 

```
Cloud.connected
Cloud.connect
Cloud.disconnect
Cloud.publish
Cloud.subscribe
Cloud.variable
Cloud.function
Cloud.syncTime
```

Perhaps `Cloud` is more descriptive of that group of functions?

cc: @jenesaisdiq

jenesaisdiq, 2015-08-10:
Welcome back! `Cloud`'s certainly more precise, but `Particle` is better for the brand. It's powerful when the functions that do magical things all start with us, rather than being a generic cloud that could be someone else. It's the killer demo: here's blink an LED... and here's blink an LED and tell the Internet, in one line of code.

zsup, 2015-08-11:
I go back on `Cloud` vs. `Particle`. `Particle` is brand-ier but `Cloud` is more descriptive. But I suppose at the end of the day `Particle` helps distinguish these functions from similar ones that might work on a different cloud, especially in cases where a device might be connecting to multiple cloud services (something we're starting to see more of). So `Particle` it is?

BracWebb, 2015-08-11:
Not that you want an outside opinion haha, but I'm always a big fan of naming parties.. At my previous job, we had an API, SCPI test functions, and a few large internal classes that were named after the company or product lines at those given times, so there were many different names. Sometimes it worked out well because it was a unique name that now just referred to that "thing", but sometimes it was frustrating. Anyway, not that you might change your name again or get acquired ;), but what if `Spark` became the name of your cloud platform/services/function, or maybe even better, another name that was part of your new brand, such as naming your cloud `Quantum` ... Particle's Quantum Cloud Platform ... What do you think?

rei-vilo, 2015-08-11:
Back to the roadmap, any date for the availability of the command line tools for the new Photon?

m-mcgowan, 2015-08-11:
particle-cli already supports the photon. and the build system is in the develop branch of the firmware repo. so afaik the photon is fully supported. are there other command line tools you're thinking of?

rei-vilo, 2015-08-12:
Thanks! I was expecting some kind of announcement. Now, does the `develop` branch mean this is still a beta release?

m-mcgowan, 2015-08-13:
I've added both `Particle` and `Cloud` aliases - when the name to use is confirmed I'll remove the other.

m-mcgowan, 2015-08-13:
@zsup @jenesaisdiq would love to get a clear direction on which name to use before the 0.4.4 release. Thanks!

zsup, 2015-08-13:
Particle!

m-mcgowan, 2015-08-23:
Here's an outline of our roadmap for migrating from Spark to Particle APIs:

- 0.4.5:  `Particle` is available alongside `Spark` - no deprecation warnings. This is when the 0.4.x line is first made available for the Core so many more users will be coming to this release. Deprecating APIs would only confuse the migration from 0.3.4 to 0.4.5. 
- 0.4.6: Add deprecation warnings to `Spark` APIs to encourage users to move over. These will be continued to be supported for at least 6 months. 
- 0.5.0: Remove the deprecated Spark APIs.

m-mcgowan, 2015-08-30:
With the roadmap is clear, I'm going to close out this issue.

closed 2015-08-30 by m-mcgowan