particle-iot/device-os #435: Photon fails on WPA2+WEP

alexmerz, 2015-06-01:
I bought a Photon at the Maker Faire Bay Area 2015. The Photon was not able to get a connection to my wlan router. The network of the router itself was visible to the Photon according to "particle setup" and "particle serial wifi". I tried several WPA2 options including WP2 mixed while setting up the Photon, and also I did an update to the latest Photon firmware.

After many attempts I was able to get it running: The router was configured to allow access via "WPA2+WEP". After I reconfigured it to WPA2 PSK only, and a setting up the Photon to "WPA2" everything worked.

The router is a re-labeled Astoria Networks Easybox 904 LTE.

abeermat, 2015-06-02:
Hi there! Thanks for the bug report!  I'm not familiar with WPA2+WEP - that seems like a very unusual (and what I thought was impossible!) configuration!

Glad to hear that WPA2+PSK works for you!

m-mcgowan, 2015-06-02:
Hi there! Thanks for the bug report!  I'm not familiar with WPA2+WEP - that seems like a very unusual (and what I thought was impossible!) configuration!

Glad to hear that WPA2+PSK works for you!

closed 2015-07-16 by m-mcgowan