particle-iot/device-os #437: incorrect return value for getPinMode()

kennethlimcp, 2015-06-02:
Example code:

```
/*
 * @brief Returns the mode of the selected pin as type PinMode
 *
 * OUTPUT = 0
 * INPUT = 1
 * INPUT_PULLUP = 2
 * INPUT_PULLDOWN = 3
 * AF_OUTPUT_PUSHPULL = 4
 * AF_OUTPUT_DRAIN = 5
 * AN_INPUT = 6
 * AN_OUTPUT = 7
 * PIN_MODE_NONE = 255  
 */
 
 
void setup(){
    pinMode(D0,OUTPUT);
    pinMode(D1,INPUT);
    pinMode(D2,INPUT_PULLUP);
    pinMode(D3,INPUT_PULLDOWN);
    pinMode(D4,AF_OUTPUT_PUSHPULL);
    pinMode(D5,AF_OUTPUT_DRAIN);
    pinMode(D6,AN_INPUT);
    pinMode(D7,AN_OUTPUT);
    pinMode(A0,PIN_MODE_NONE);
    
    Serial.begin(9600);
    while(!Serial.available()) Spark.process();
    Serial.println(getPinMode(D0));
    Serial.println(getPinMode(D1));
    Serial.println(getPinMode(D2));
    Serial.println(getPinMode(D3));
    Serial.println(getPinMode(D4));
    Serial.println(getPinMode(D5));
    Serial.println(getPinMode(D6));
    Serial.println(getPinMode(D7));
    Serial.println(getPinMode(A0));
}
```

Output: 
```
1
0
2
3
4
5
6
7
0
```

m-mcgowan, 2015-08-17:
It's not obvious to me how the actual behavior differs from the expected behavior. @kennethlimcp  could you say more about how you think this should work?

`PIN_MODE_NONE` is not a valid mode, so that's the only difference I can see, you might expect the result to be 255, but it's not since the mode was never set.

kennethlimcp, 2015-08-17:
See that for D0, the reading is 1 which should be 0 for OUTPUT and for D1 the reading is 0 which should be 1 for INPUT

m-mcgowan, 2015-08-17:
Here's the PinMode enum

```
typedef enum PinMode {
  INPUT,
  OUTPUT,
  INPUT_PULLUP,
  INPUT_PULLDOWN,
  AF_OUTPUT_PUSHPULL, //Used internally for Alternate Function Output PushPull(TIM, UART, SPI etc)
  AF_OUTPUT_DRAIN,    //Used internally for Alternate Function Output Drain(I2C etc). External pullup resistors required.
  AN_INPUT,           //Used internally for ADC Input
  AN_OUTPUT,          //Used internally for DAC Output
  PIN_MODE_NONE=0xFF
} PinMode;
```

https://github.com/spark/firmware/blob/develop/hal/inc/pinmap_hal.h#L42-L52

`OUTPUT` is 1 and `INPUT` is 0, so the result above is perfectly correct afact.

m-mcgowan, 2015-08-22:
Closing this since I think the output is correct. @kennethlimcp if I'm overlooking something feel free to re-open and explain where I'm going wrong. Thanks!

closed 2015-08-22 by m-mcgowan