particle-iot/device-os #439: Why dont you use LibWdi to create the DFU Driver for Windows ?

rogerclarkmelbourne, 2015-06-03:
I've used https://github.com/pbatard/libwdi on other projects as it simplifies driver creation

I presume that the DFU driver isnt really a driver but just an association between the USB VID/PID with the Windows DFU driver, so LibWDI should work fine for this ??

m-mcgowan, 2015-06-03:
Thanks for the tip!  Does this mean users would no longer have to install Zadig?

rogerclarkmelbourne, 2015-06-03:
If you use the standard Windows DFU driver, then there is no need to use Zadig, 

I use it in my repo for STM32 DFU and it works on W7, W8 and W8.1 (no idea if it will continue to work on W10, but according to pbatard's repo readme it works with W10

I didn't write it myself, but someone contributed this to my repo

https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/drivers/win/src

There are instructions on how to use pbatards repo then patch it, 

Actually, I'm not even sure you need to do that, 

If you just grab the exe from my repo and then use this command in a batch file, it seems to install just fine.

```
echo Installing Photon DFU driver...
"%~dp0wdi-simple" --vid 0x2b04 --pid 0xD006 --type 1 --name "Photon DFU Mode" --dest "%~dp0photon-dfu"
echo.
``

I just ran dfu-util.exe -l and I see

dfu-util 0.7-msvc

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Found DFU: [2b04:d006] ver=0200, devnum=5, cfg=1, intf=0, alt=1, name="UNKNOWN", serial="UNKNOWN"
Found DFU: [2b04:d006] ver=0200, devnum=5, cfg=1, intf=0, alt=0, name="UNKNOWN", serial="UNKNOWN"

Which implies it can see the Photon

But I've not tried to flash new firmware to it

Actually, I'm interested in flashing completely different firmware to the board, i,e directly compile in the Arduino and then flash via DFU using the Arduino IDE

(Albeit, I don't have support for the ST32F2 at the moment, but i have F1 F3 and F4, so I should be able to get F2 working if I put my mind to it)

m-mcgowan, 2015-09-10:
Thanks for this. Building/deploying Windows drivers isn't something I'm currently familiar with, so I've a few questions:

- The name and serial attributes are reported as UNKNOWN above. That doesn't seem right since the name is used to convey the memory map to the dfu host. 

- Can l sign the driver using my own cert? A brief scan of the readme of libwdi mentioned at uses an automatically generated certificate.

rogerclarkmelbourne, 2015-09-10:
Sorry, but I'm not a libwdi expert, I just use it on other projects that need usb drivers.

You should probably direct technical questions to the libwdi team

m-mcgowan, 2015-09-14:
No probs, and thanks for the suggestion. To answer your question about why we don't use libwdi, it's because we don't really have the expertise to build this, nor the time, given our other priorities at present.  The current solution (Zadig) has been adequate for our needs so far so we'll continue with that until we see a need to change. 

Much appreciative of the info :+1:  - I didn't know the library existed, so will come back to this if we need something other than Zadig.

closed 2015-09-14 by m-mcgowan