particle-iot/device-os #440: POSIX Time is not set starting from definition

kennethlimcp, 2015-06-03:
The definition is:

```
since 00:00:00 Coordinated Universal Time (UTC), Thursday, 1 January 1970,
```

But on power up, the **photon** says: `Thu May 8 20:20:xx 1913`

Haven't looked into the firmware but it should at least follow the POSIX standard for starting point. 1913 before POSIX time is defined.

m-mcgowan, 2015-06-03:
@satishgn could you look into this please - I imagine it's just a trivial initialization that's missing.

satishgn, 2015-06-05:
I saw this problem now on my photon too. Also noted that the RTC is reset everytime on system start to the non-standard unix time. Ideally after syncing the new time from cloud, it should retain the time-keeping even after reset with power applied but it is not doing so like on Core. Even waking from sleep after standby would be problematic in this scenario. There's something that went wrong since I tested this last time. I suppose Wiced is reseting/manipulating the time everytime after reset. Looking into this issue now...

satishgn, 2015-06-05:
Ok Confirmed that WICED is resetting the default RTC time everytime at startup:
/* Default RTC time. Set to 12:20:30 08/04/2013 */
static const platform_rtc_time_t default_rtc_time =
{
   .sec     = 30,
   .min     = 20,
   .hr      = 12,
   .weekday = 1,
   .date    = 8,
   .month   = 4,
   .year    = 13,
};
Note: Here arduino month is default_rtc_time.month+1

Probably this happened after updating to latest WICED version. In the older version, The WICED-RTC feature was disabled.

m-mcgowan, 2015-06-12:
I'm just not sure this is a bug. If the time is reset after waking from sleep, that would be an issue, but how is having a default of 2013 worse than a default of 1970 - both are wrong.

kennethlimcp, 2015-06-13:
My use case for an LCD:

```
void setup(){
if(Time.year() == xxxx){
        LCD_PORT.write("Time not synced.");
     }
}
```

I had to figure out the hard way that the new codebase is not following the Arduino or rather POSIX Time standard.

It's not a bug but a transition for people would result in them spending hours figuring this out without it being stated as the behavior.

See definition here: http://playground.arduino.cc/Code/DateTime

closed 2015-06-13 by kennethlimcp

m-mcgowan, 2015-06-13:
That's a fair point (and one I had in the back of my mind.) I'll reset the WICED default to 1970.

satishgn, 2015-06-13:
When I last tested the working System Time/RTC module :  RTC code in WICED was disabled and we were using our own RTC configuration functions. Now I think WICED is resetting the time to its default WICED time everytime at startup even with +3V3 power available during reset. If WICED RTC code is disabled, it should solve the problem right away for wrong posix time and sleep functionality.

towynlin, 2015-06-18:
Reminder to post a resolution here: https://community.particle.io/t/time-is-different-on-the-photon-vs-core/12322

satishgn, 2015-06-18:
Sure. Debugging this one right now...

satishgn, 2015-06-19:
Commented the RTC init code in the WICED library to fix this issue.