particle-iot/device-os #443: attachInterrupt issue on Photon

zsup, 2015-06-05:
I'm having trouble setting up an interrupt on a Photon; would love some help determining if this is an issue or if I am simply doing something wrong.

The hardware setup for the sample code below is that I have `A0` and `D0` connected through a switch. `A0` is pulled high, and `D0` is pulled low through a 10K resistor. I expect to see a rising edge when the switch is closed, and falling edge when the switch is opened.

Here's my code:
```
void setup() {
    pinMode(A0, OUTPUT);
    pinMode(D7, OUTPUT);
    digitalWrite(A0, HIGH);
    attachInterrupt(D0, trigger, CHANGE);
}

void loop() {
}

void trigger() {
    digitalWrite(D7, HIGH);
}
```

Despite flipping the switch multiple times, `trigger()` is never called and the LED on `D7` never turns on. The only explanation I can come up with is that the interrupt is not being set correctly.

Note that the hardware has been validated; if I do the same thing by using `digitalRead()` instead of an interrupt, it works as expected.

zsup, 2015-06-05:
As a follow up, here is the code that works as expected (not interrupt driven, LED turns on when the switch is closed):

```cpp
int trigger = 0;

void setup() {
    pinMode(A0, OUTPUT);
    pinMode(D7, OUTPUT);
    pinMode(D0, INPUT);
    digitalWrite(A0, HIGH);
}

void loop() {
    trigger = digitalRead(D0);
    digitalWrite(D7, trigger);
}
```

kennethlimcp, 2015-06-06:
Tested that the example above is not triggering the interrupt function.

technobly, 2015-06-07:
We previously had a hard fault happening... @m-mcgowan was the attachInterrupts() working as expected after the resolution of the hard fault?
https://github.com/spark/firmware-private/issues/98

mojtabacazi, 2015-06-09:
FYI, ADC_IRQHandler won't work either. Seems like `Default_Handler` is being called

mojtabacazi, 2015-06-11:
My dirty solution to get ADC_IRQHandler to work

```c
void ADC_IRQHandler () {
    volatile uint16_t sampleVal = ADC_GetConversionValue(ADC1);
    // Process sampleVal ...
    ADC_ClearITPendingBit(ADC1, ADC_IT_EOC);
}

setup() {
    uint32_t *p = (uint32_t *)SCB->VTOR;
    *(p+34) = (uint32_t)ADC_IRQHandler;
    // Now you can setup ADC and enable IRQ
} 

loop () {
    // Nothing
}
```

m-mcgowan, 2015-06-11:
:+1: That's a nice (dirty ;-P) hack! We will be providing an officially supported API in the next release for hooking interrupts so you don't have to resort to these levels of hackery!

steelydev, 2015-06-13:
No pressure.  But any estimate when interrupts will work on the Photon?

m-mcgowan, 2015-06-15:
I just tested this - interrupts work for pins D1, D2, D3 (that was the scope of my test), but not D0. My understanding is that this is a specific issue with triggering an interrupt on D0.

steelydev, 2015-06-15:
We actually had our PCB printed using A5.  Any estimate there?

m-mcgowan, 2015-06-15:
The issue here was for D0. Have you seen that A5 doesn't work also?

steelydev, 2015-06-15:
@m-mcgowan Went through them all

A0 - doesn't work
A1 - doesn't work
A2 - works
A3 - works
A4 - works
A5 - doesn't work
A6 - SOS
WK - works
RX - works
TX - doesn't work
D0 - doesn't work
D1 - works
D2 - works
D3 - works
D4 - works
D5 - doesn't work
D6 - doesn't work
D7 - doesn't work

Code used:
```cpp 
volatile bool interrupt = false;
static int INTERRUPT_PIN = TX;
static int LED = D0;

void setup() {
  pinMode(INTERRUPT_PIN,INPUT_PULLUP);
  attachInterrupt(INTERRUPT_PIN,ISR,FALLING);
}

void loop() {
  if (interrupt) {
    blinkLED();
    interrupt = false;
  }
}

void ISR() {
  interrupt = true;
}

void blinkLED() {
  pinMode(LED,OUTPUT);
  digitalWrite(LED, HIGH);
  delay(50);
  digitalWrite(LED, LOW);
  delay(50);
}
```

m-mcgowan, 2015-06-15:
Thanks very much for this. @satishgn is most familiar with the low-level code - he's able to look at this on wednesday/thursday this week.

satishgn, 2015-06-23:
On Photon: Following are the shared External Interrupt (EXTI) lines so only one can work at a time:

Mode Button, D0, A5 (D0, A5 doesn’t work at present)
* D1, A4
* D2, A0, A3
* D3, DAC
* D4, A1

Except D0 and A5, all other pins that @steelydev mentioned above should work now with the latest "develop" branch commits.

closed 2015-06-23 by satishgn

christophevg, 2017-01-11:
Just to add a note here for reference if others pass by with the same problem:
In the documentation on System Modes ([https://docs.particle.io/support/troubleshooting/mode-switching/photon/#semi-automatic-mode](https://docs.particle.io/support/troubleshooting/mode-switching/photon/#semi-automatic-mode)), an interrupt on `D0` is used. As `D0` doesn't seem to work, this example also cannot work and can cause quiet some searching and includes a risk ;-)