particle-iot/device-os #453: Function to list the SSIDs

Rockvole, 2015-06-14:
Can we create a function on the Photon to return the list of SSIDs which the Photon has available ?

An example would be :

Wifi.getCredentials() or Wifi.getSsids()

returns an array of strings with the SSIDs in it.

This was discussed on the forum :
https://community.particle.io/t/manage-wifi-credentials/9982

Thank-you

m-mcgowan, 2015-06-14:
Can you say a little about how you plan to use this? I'm having difficulty envisaging what the use case might be. 

Returning arrays are tricky in C - they have to be allocated somehow, and dynamic allocation isn't usually a good idea. What we can do is have the caller provide the array for the result, and the function populates this. 

```
struct WiFiAccessPoint {
   char SSID[33]; 
   uint8_t BSSID[6];  
   SecurityType securityType;
   uint8_t channel;
   int RSSI;  // when scanning
};

/**
 * Retrieves the configured access points that this device will attempt to connect to.
 * @param results: the buffer of WiFiAccessPoint instances to populate. Can be NULL
    in which case the function returns the number of access points available.
 * @param results_size: the number of items in the `results` buffer
 * @return The total number of APs configured. 
 */
int WiFi.configuredAP(WiFiAccessPoint* results, size_t results_size);

```

I'm confident we can build this on the Photon since we have access to the configured WiFi APs, but I'm less certain about the Core since the details are stored inside the CC3000. Any insight on how to implement for the Core welcomed!

There's also been talk on the forum of an API to allow networks to be scanned. The `WiFiAccessPoint` struct can be reused for that:

```
/**
 * Scans for local access points. 
 * @param results A buffer to store the results. Can be NULL in which case the function returns
   the number of APs available. 
 * @param results_size the number of elements in the buffer.
 * @return The number of APs found. Up to `results_size` APs will be written to the `results` buffer.
int WiFi.scanAP(WiFiAccessPoint* results, size_t results_size);
```

Rockvole, 2015-06-14:
I'm no C expert so whatever struct / array you think is best is fine with me.

I put in a couple of use cases on the forum :
https://community.particle.io/t/manage-wifi-credentials/9982/11

Rockvole, 2015-06-14:
There could be further cases requiring managing the list of ssids.

If you have 1 main ssid which you never want to lose, when you keep
adding new ssids your main one will eventually drop off the end and
you might not realize. I'm not sure whether this idea could be extended
to prevent that.

I guess my use cases could be considered debugging, not really production.
There may be some production use case where you need a device to
have a large number of ssids stored and you need to see what is stored.
Such as a device where you take it to various locations to take a reading
or to send messages when you reach those locations.
Such as starbucks wifi, google wifi, work wifi, home wifi etc etc.

Once you start having larger numbers it becomes useful to be able to see
what connections you have so you can manage this list.

Rockvole, 2015-06-14:
Another situation I came across recently where a number of ssids
were required. I put my sensors in someones house and he had high readings.
He lives in a multi-unit complex with 6 units. He had high readings and
then wanted to pass the device to adjacent neighbors to see what their
readings were, so then we started adding more connections, but still wanted 
to be able to return the device to the original unit. In this case we never
exhausted the number of connections but I imagine it could go over 10
fairly easily in this way.

chybis, 2015-07-14:
If it is possible, I would recommend fetching just one AP details at the time. In crowded areas you can see tens of WiFi APs, which might be too much for a Core/Photon to keep in memory.

Use case is to have full AP data (SSID, ...) available only when a GUI is constructed or refreshed, but otherwise forget all data except some key until needed.

Let's say something like this:

```
/**
 * Returns count of available local access points. 
 * @return The number of APs found.
 */
int WiFi.scanCountAP();

/**
 * Stores a scanned AP into a supplied buffer. 
 * @param result A buffer to store the scanned AP data.
 * @param index Index of found AP, valid range is from 0 to WiFi.scanCountAP() - 1.
 * @return True if the data was written, false if no such index exists.
 */
bool WiFi.scanAP(WiFiAccessPoint result, int index);
```

It requires the scanned list doesn't change between calls [much]. Or user can compute a hash somehow and just iterate through `WiFi.scanAP()` until it returns false, comparing hashes to avoid duplicate results.

WIFI library may assist with this, either returning hash with the AP data, or perhaps working with hashes internally. For example:

```
/**
 * Scans for local access points. 
 * @param result_hashes A buffer to store the result hashes. Can be NULL in which case
   the function returns the number of APs available. 
 * @param results_size the number of elements in the buffer.
 * @return The number of APs found. Up to `results_size` AP hashes will be written
   to the `result_hashes` buffer.
 */
int WiFi.scanAP(uint_64t* result_hashes, size_t results_size);

/**
 * Stores a scanned AP into a supplied buffer. 
 * @param result A buffer to store the scanned AP data.
 * @param result_hash Hash of found AP, as returned by WiFi.scanAP().
 * @return True if the data was written, false if no such hash exists.
 */
bool WiFi.getAPData(WiFiAccessPoint result, uint_64t result_hash);
```

Both functions may walk through the AP list (as returned by the WiFi chip), calculating hashes on the fly. `scanAP()` just returns the hashes and forgets everything. `getAPData()` scans the list again, computing hashes, stopping when found AP matches the supplied hash, storing it to supplied bufer.

This is obviously trade of time for memory. I don't know how fast the scan is, it might be unreasonable to wait 5 seconds for single call to `getAPData()` when iterating over 10 networks.

Rockvole, 2015-07-14:
I guess there is some confusion about what I am talking about.
I want a function to list the ssid's currently stored in the Photon.

There have been discussions on the forum regarding having a webpage
on the Photon to allow the managing of wifi connections:
https://community.particle.io/t/photon-softap-usage/11035/42

I guess this would come under that umbrella. For that use case, full
management of ssids would be needed, such as :
1) List all known ssids. 
2) Delete a particular ssid. 
3) Replace an ssid
4) Add a new ssid + password
5) Set connection order, if there are multiple networks available
6) Connect to a particular network now

pingdynasty, 2015-08-13:
Personally I would much prefer, as a start:
- `bool WiFi.configureAccessPoint(const char *ssid, const char *password, unsigned long security)` : set up a local soft AP that other devices can connect to
- some method of scanning and retrieving data about local APs

The app can then itself implement whatever webserver, websockets, DNS or zero-conf services it wishes.
Different template apps can be developed for different purposes, without making the API and library horrendously complicated.

Rockvole, 2015-08-13:
Yes that is the basic minimum which would be needed. I think it would be useful to have more functionality. At the moment you get 8 connections stored (6 on the Photon) and you can keep adding connections until the first one drops off the end. I don't believe there is currently a way to set an ssid which wont drop off the end.

A webpage like this on the Photon would need additional API calls :
https://community.particle.io/t/webpages-on-photon-softap/13314/3

m-mcgowan, 2015-09-13:
Managing the on-device SSIDs is clearly a useful function. I think where possible this should be added to the serial setup and SoftAP interfaces too.

PR #759 (2016-01-14): Implement WiFi.getCredentials
* User provides an array of WiFiAccessPoint that get filled with the APs saved on the device
* Returns number of APs on device
* Photon only. Returns 0 on the Core since it cannot read APs back from the CC3000 :disappointed:

Changes
* Implement wlan_get_credentials in HAL
* Implement WiFi.getCredentials with array arguments
* Add API test

Fixes #453

Usage:
```
    WiFiAccessPoint ap[5];
    int found = WiFi.getCredentials(ap, 5);
    for (int i=0; i<found; i++) {
        Serial.print("ssid: ");
        Serial.println(ap[i].ssid);
        Serial.println(ap[i].se [... cut]
commit 06fe2835 (2016-01-14): Implement WiFi.getCredentials

* User provides an array of WiFiAccessPoint that get filled with the APs
  saved on the device
* Returns number of APs on device
* Photon only. Returns 0 on the Core since it cannot read APs back from the CC3000

Changes
* Implement wlan_get_credentials in HAL
* Implement WiFi.getCredentials with array arguments
* Add API test

Fixes #453
closed 2016-01-14 by m-mcgowan