particle-iot/device-os #454: Can set SPI clock frequency in a platform-neutral way

m-mcgowan, 2015-06-15:
https://community.particle.io/t/system-clock-of-photon-vs-core-affecting-spi/12571/3?u=mdma

```
SPI.setClockFrequency(24, MHz);
```

The API would choose the clock divider that results in the highest frequency not higher than the frequency specified.

technobly, 2015-07-01:
Additionally, perhaps we can specify something that tries to normalize to SPI speeds to arduino?  I feel like if you are coming from the angle of porting an ARDUINO library and you already have to convert SPI_CLOCK_DIV4 to 4MHz and start calculating what frequencies you can achieve with Core or Photon, you might as well stay in control and set the frequency as close to the desired frequency as possible.  So maybe by adding a normalize() method, we can do this automatically.

```cpp
SPI.normalize(SPI_ARDUINO):
SPI.setClockFrequency(SPI_CLOCK_DIV4);  // normalizes this to approx. 16/4MHz 
// which is 120/4 on the photon SPI_CLOCK_DIV32 = 3.75MHz
// and is 72/4 on the core SPI_CLOCK_DIV16 = 4.5MHz
// and without normalize() would be 30MHz on the Photon and 18MHz on the Core.
```

m-mcgowan, 2015-07-15:
This is a good idea and will make porting libraries to both the Core, Photon and Electron (from Arduino) much easier. 

cc: @jenesaisdiq We can find a more suitable name since normalize is quite generic. (Normalize what?) Something like `SPI.setClockReference(ARDUINO)` .  Then library porters only need to include:

```
#ifdef SPARK
SPI.setClockReference(ARDUINO);
#endif
```

At the start of their library to have all the other SPI functions work unmodified.

m-mcgowan, 2015-08-23:
Added a candidate implementation. Happy to rework function names if we find more suitable alternatives.

closed 2015-09-02 by m-mcgowan