particle-iot/device-os #455: Time.timeStr() should support alternative time format

ScruffR, 2015-06-16:
While it might be useful for US applications and acceptable for IT people of other nationality, the default `timeStr()` format "DDD MMMM DD hh:mm:ss YYYY" might not appeal to the global crowd of potential standard consumers ;-)

Hence it might be good to have either an additional static field in the `Time` class to provide a desired format to define the localized behaviour of `timeStr()` or/and an overload of `timeStr(const char* timeFormat)` to set the format on the fly.
This might also call for the need to supply localized day and month names (short and long) and the addition of `hh24` or `HH` as a 24 hour display and a place holder for `AM/PM`. 
An format descriptor to display timezone might also be good.

The default date time format might also be changed into something more globally used (e.g. XML time code `2002-05-30T09:30:10Z` or `2002-05-30T09:30:10-02:00`).

m-mcgowan, 2015-06-16:
:+1: for the global, standardized time format. Localizing might take up quite a bit of memory and is probably best done externally.

ScruffR, 2015-06-16:
Thanks, this is how I'm doing it at the moment, but thought it might be cool if it was on board ;-)
But I'd definetly appreciate a more globally "useful" default.

ScruffR, 2015-08-22:
Hi Mat, just for reference http://www.w3.org/TR/NOTE-datetime

m-mcgowan, 2015-08-22:
I've added the formatting mechanism that supports the existing asctime() format and ISO 8601. Feel free to add other formats as you need them.

ScruffR, 2015-08-22:
I'll have a look at yours, but this was what I came up with
```
String TimeClass::timeStr(time_t t)
{
        struct tm *calendar_time;
	char iso8601wSec[26];        // date time: YYYY-MM-DDTHH:mm:ss
	char iso8601tz[7] = { "Z" }; // time zone offset: Z\0 or +hh:mm
	
	if (time_zone_cache)
	{
		t += time_zone_cache;
		sprintf(iso8601tz, "%+02d:%02d", 
                        time_zone_cache / 3600, 
                        time_zone_cache / 60);
	}

	calendar_time = localtime(&t);
	strftime(iso8601wSec, sizeof(iso8601wSec), "%FT%T", calendar_time);
	strcat(iso8601wSec, iso8601tz);
	String calendar_time_string = String(iso8601wSec);

	return calendar_time_string;
}
```

m-mcgowan, 2015-08-30:
`Time.setFormat(const char*)` - I like that idea! But I'd prefer not to have a fixed size buffer. Instead we can simply make it a `const char*` pointer and document that users should only pass in string literals, or a pointer to a globally allocated buffer. 

E.g.

```
void setup() {
   Time.setFormat("%H:%M");
}
```

ScruffR, 2015-08-30:
I thought of that at first, but couldn't convince myself that users would read that doc and so they would soon complain that their strings are not formatted right ;-)

Then I thought of `String` objects, but I don't really like them too.

But who cares, we have to tell users over and over not to pass a char array as `&arr` so we can accept correct the few who get wrong time strings ;-)

m-mcgowan, 2015-08-30:
I have to rework the API a bit for the core. `strtfime` takes about 8K of flash memory, so we really have to only include it if the user is doing non-default formatting.

m-mcgowan, 2015-09-02:
Implemented suggestion to make formatting string configurable by the caller. With unit tests.

closed 2015-09-02 by m-mcgowan