particle-iot/device-os #457: [photon] strdup() returns nonsense

m-mcgowan, 2015-06-16:
`strdup()` is not returning a copy of the given string

Test code:

void loop()
{
    RGB.control(true);
    RGB.color(255,255,0);
    delay(1000);
    Serial.println("Hello world!");
    String list = "1,2,3,4,5";
    char* dup = strdup(list.c_str());
    if (strcmp(dup, list.c_str())) {
        Serial.println(dup);
        Serial.println("duplicate doesn't match");
    }
    char* tok = strtok(dup,",");
    while (tok) {
        Serial.println(tok);
        tok = strtok(NULL, ",");
    }
    delay(1000);
    RGB.color(255,0,255);

    delay(1000);
}
```

This should print 1,2,3,4,5 on separate lines to USB serial, but prints random characters.

m-mcgowan, 2015-06-16:
## Workaround

Add this to the application code:

```

char* strdup(const char* s)
{
    if (!s)
        return NULL;
    int len = strlen(s);
    char* result = (char*)malloc(len+1);
    strcpy(result, s);
    return result;
}
```

mtnbrit, 2015-07-20:
I'm running 0.4.3 and strdup is still broken, thanks for your workaround in the meantime

m-mcgowan, 2015-07-20:
Thanks. The fix was in the application layer which will be released soon to the online IDE.

mtnbrit, 2015-07-20:
does that include Particle Dev?

m-mcgowan, 2015-07-20:
Ah yes, I could have been clearer.. Particle Dev and the WebIDE share the same cloud compile service.

m-mcgowan, 2015-08-13:
This is now fixed by the malloc_r fix #529, since `strdup()` is using `malloc_r` behind the scenes to allocate memory. Previously, malloc_r was not redirected to the system module, and instead was implemented by importing runtime functions into the user module. The user module had a dummy _sbrk() implementation that never advanced the heap and returned NULL, so no additional memory was ever allocated. 

I didn't check to confirm, but I suspect `dup` in the test case would have been NULL, which doesn't seem to cause a crash on the photon at present since bus errors are not trapped. When using memory allocation functions, it's good practice to check the return value, which will be NULL when memory is exhausted.

closed 2015-08-13 by m-mcgowan