particle-iot/device-os #459: Core compile is showing undefined reference to `socket_create_tcp_server'

thomast74, 2015-06-18:
/tmp/cca62r7B.ltrans21.ltrans.o: In function `TCPServer::begin()':
/home/thomast74/Projects/spark-firmware/main/src/spark_wiring_tcpserver.cpp:44: undefined reference to `socket_create_tcp_server'

Looking at the source code for /hal/src/core/socket_hal.c the method socket_create_tcp_server is not implemented but socket_hal.h requires it. 
The socket_hal.c file includes a socket_create_nonblocking_server but this method has a different signature.

m-mcgowan, 2015-07-12:
Fixed in the commit above. @thomast74 I'd be grateful if you could please confirm the fix so I can close this issue.

thomast74, 2015-07-15:
Compile works and application with these changes works as well. I think problem is resolved.

Thanks

m-mcgowan, 2015-07-15:
Thanks for getting back.

closed 2015-07-15 by m-mcgowan