particle-iot/device-os #460: I2C call from interrupt

maghis, 2015-06-21:
Communicating with I2C from an ISR (timer in my case) causes the core to reset, especially when the I2C device is not present or fails to respond.

The issue is due to the way Wire is implemented.
The current implementation handles timeouts using millis() which is incremented using an ISR (void SysTick_Handler(void) that calls System1MsTick()): millis() doesn't increment when called in a ISR, this leads to an endless loop instead of a timeout when waiting for I2C events.

Switching to micros() should do the trick, since those are managed by the chip and just derived form the clock cycle counter (DWT->CYCCNT).

m-mcgowan, 2015-06-21:
Great you found a fix. We will change the call to micros(). But are you sure that calling ISR functions from an interrupt is a good idea? I would instead  suggest setting a flag in the ISR which you then check in the main loop.

m-mcgowan, 2015-06-21:
@maghis - Could you try the fix above and confirm if it's working for you? Thanks!

maghis, 2015-06-21:
@m-mcgowan I am sampling from an I2C adc stuff that is kind of time critical (even if slow, about 100Hz). But I agree, if I had an os I would have used a separate thread with a simple sleep().

I took a look at the code and it looks exactly like my fix, it should work.
Cheers!

Is there a way to contribute directly?

m-mcgowan, 2015-06-21:
Thanks for checking. Yep you can contribute by submitting a [pull request](https://help.github.com/articles/using-pull-requests/).

m-mcgowan, 2015-07-16:
Confirmed fixed in the comments above.

closed 2015-07-16 by m-mcgowan