particle-iot/device-os #464: Spark.syncTime() cannot compile in Particle Dev

pomplesiegel, 2015-06-24:
Within Particle Dev, a call to Spark.syncTime() results in error `undefined reference to spark_protocol_send_time_request`.

technobly, 2015-07-01:
@pomplesiegel thanks for the bug report!  I verified this by compiling the following app for a Photon in the online IDE.  I also noticed another potential issue with Spark.unsubscribe() which is not documented yet.  This app reveals both errors:

```cpp
#define ONE_DAY_MILLIS (24 * 60 * 60 * 1000)
unsigned long lastSync = millis();

void loop() {
  if (millis() - lastSync > ONE_DAY_MILLIS) {
    // Request time synchronization from the Particle Cloud
    Spark.subscribe("my-event", myHandler);
    Spark.unsubscribe();
    Spark.syncTime();
    lastSync = millis();
  }
}

void myHandler(const char *event, const char *data)
{
}
```

### Error output:
```
../../../build/target/user/platform-6/libuser.a(spark.synctime.o): In function `CloudClass::unsubscribe()':
/spark/compile_service/shared/workspace/6_hal_12_0/firmware-private/user/../wiring/inc/spark_wiring_cloud.h:98: undefined reference to `spark_protocol_remove_event_handlers'
../../../build/target/user/platform-6/libuser.a(spark.synctime.o): In function `CloudClass::syncTime()':
/spark/compile_service/shared/workspace/6_hal_12_0/firmware-private/user/../wiring/inc/spark_wiring_cloud.h:103: undefined reference to `spark_protocol_send_time_request'
collect2: error: ld returned 1 exit status
make: *** [4537e0f2057acd962a59c421e734154baa4a2816cda29e8b42b19c51a16a.elf] Error 1
```

#### Ref to dynalib entries:
https://github.com/spark/firmware/blob/develop/communication/src/communication_dynalib.h

m-mcgowan, 2015-07-13:
This is fixed in 0.4.1, just waiting for that to be released. (This issue duplicates https://github.com/spark/firmware/issues/426)

closed 2015-07-13 by m-mcgowan

technobly, 2015-07-14:
@m-mcgowan did you see the mention above about Spark.unsubscribe(); as well.  Has this also been added in 0.4.1?

m-mcgowan, 2015-07-14:
Spark.unsubscribe() isn't part of our public API yet - I'll add it when the docs are in place. (The primary motivator for that function is for testing.)