particle-iot/device-os #468: receiving UDP on Photon not working

steelydev, 2015-06-27:
Is anyone aware that this simple program does work on Core but not on Photon?

```cpp
// UDP Port used for two way communication
unsigned int localPort = 8888;
int i = 0;
// An UDP instance to let us send and receive packets over UDP
UDP Udp;

void setup() {
  // start the UDP
  Udp.begin(localPort);

  // Print your device IP Address via serial
  Serial.begin(9600);
  Serial.println(WiFi.localIP());
}

void loop() {
  // Check if data has been received
  i = Udp.parsePacket();
  Serial.print("parsePacket returns ");Serial.println(i);

  if (i > 0) {

    // Read first char of data received
    char c = Udp.read();

    // Ignore other chars
    Udp.flush();

    // Store sender ip and port
    IPAddress ipAddress = Udp.remoteIP();
    int port = Udp.remotePort();

    // Echo back data to sender
    Serial.print("echo ");Serial.println(c);

    Udp.beginPacket(ipAddress, localPort+1);
    Udp.write(c);
    Udp.endPacket();
  }
  delay(500);
}
```

parsePacket aways returns 0 on Photon.

I am unable to read any UDP on Photon.

technobly, 2015-07-01:
Thanks @steelydev for this bug report.  I know that Mat has been working on some UDP updates, which may resolve this... but if not already resolved I've also captured this in our internal project management system.

I have no idea of the state of this branch, but perhaps you can try it out if you have time:
https://github.com/spark/firmware/compare/develop...feature/udp

m-mcgowan, 2015-07-11:
I'm pleased to say UDP has been fixed in 0.4.2. If you can update your system firmware to that (from https://github.com/spark/firmware/releases/tag/0.4.2) that should resolve the issue.

m-mcgowan, 2015-07-24:
@steelydev did you get a chance to try this and did it fix the UDP issue you were seeing?

steelydev, 2015-07-24:
Yes! Good! The code above now works on Photon!

closed 2015-07-25 by m-mcgowan

wmoecke, 2015-09-22:
Hi
I wrote a piece of code that listens on UDP packets, it works under certain scenarios (e.g., when it's the main code in the sketch). But when I tried adding my code into another existing piece of code (https://github.com/sparcules/Spark_Pixels/blob/master/Firmware/SparkPixels.ino) I never get anything back from parsePacket(). I wonder if there's something obvious I may have overlooked.
```c++
#include "neopixel/neopixel.h"

//NEOPIXEL Defines
#define PIXEL_CNT       512
#define PIXEL_PIN       D0
#define PIXEL_TYPE      WS2812B

#define NR_OF_PANELS    8

Adafruit_NeoPixel strip = Adafruit_NeoPixel(PIXEL_CNT, PIXEL_PIN, PIXEL_TYPE);

// An UDP instance to let us receive packets over UDP
UDP Udp;

// Define some tpm2.net constants
#define TPM2NET_LISTENING_PORT  65506
#define TPM2NET_HEADER_SIZE     6
#define TPM2NET_HEADER_IDENT    0x9C
#define TPM2NET_CMD_DATAFRAME   0xDA
#define TPM2NET_CMD_COMMAND     0xC0
#define TPM2NET_CMD_ANSWER      0xAC
#define TPM2NET_FOOTER_IDENT    0x36
#define TPM2NET_PACKET_TIMEOUT  0xFA    // 1/4 of a second

#define NR_OF_PANELS            cube.size
#define PIXELS_PER_PANEL        (PIXEL_COUNT / NR_OF_PANELS)

// 3 bytes per pixel or 24bit (RGB)
#define BPP                     3

// Package size we expect. The footer byte is not included here!
#define EXPECTED_PACKED_SIZE    (PIXELS_PER_PANEL * BPP + TPM2NET_HEADER_SIZE)

float timeLastPacketReceived;

void setup() {
    cube.begin();
    cube.background(black);
    cube.show();
    
    // Start the UDP
    if(Udp.setBuffer(EXPECTED_PACKED_SIZE))
        Udp.begin(TPM2NET_LISTENING_PORT);
}

void loop() {
    // Checks for the presence of a UDP packet, and reports the buffer size
    int32_t bytes = Udp.parsePacket();
    if(bytes >= EXPECTED_PACKED_SIZE) {
        // Read 198 bytes of data from the buffer
        char data[EXPECTED_PACKED_SIZE];
        Udp.read(data, EXPECTED_PACKED_SIZE);
        
        // Ignore all other chars
        Udp.flush();
        timeLastPacketReceived = millis();
        
        /*------------------*/
        /*-- Header check --*/
        /*------------------*/
        // Block Start Byte
        if(data[0] == TPM2NET_HEADER_IDENT) {
            // Block Type: Command Packet
            if(data[1] == TPM2NET_CMD_COMMAND)
                return; // Don't care

            // Block Type: Frame Data Packet - that's what we want
            if(data[1] == TPM2NET_CMD_DATAFRAME) {
                // Calculate frame size
                char frameSize = data[2];
                frameSize = (frameSize << 8) + data[3];
        
                // Use packetNumber to calculate offset
                char packetNumber = data[4];
                char totalPackets = data[5];
        
                // Calculate offset
                uint16_t index = packetNumber * PIXELS_PER_PANEL;
                uint16_t voxelIdx = TPM2NET_HEADER_SIZE;

                // Start drawing!!
                for(uint16_t z = NR_OF_PANELS-1; z >= 0; z--) {                  // We're only dealing in 2 dimensions (width & height)
                    for(uint16_t col = 0; col < NR_OF_PANELS; col++) {           // Linewise, starting from leftmost index
                        for(uint16_t row = NR_OF_PANELS - 1; row >= 0; row--) {  // Columnwise, starting from topmost index
                            uint16_t stripIdx = (row * PIXELS_PER_PANEL) + (z * NR_OF_PANELS) + col;
                            // Take 3 color bytes from buffer
                            strip.setPixelColor(stripIdx, strip.Color(data[voxelIdx], data[voxelIdx + 1], data[voxelIdx + 2]));   //(z, col, row, pixelColor);
                        }
                        voxelIdx+=3; // Increment buffer index by 3 bytes
                        if(stop == TRUE) {return;}
                    }
                }
                // Display!!
                if(packetNumber == totalPackets) {strip.show();}
            }
        }
    }
    else if(millis() - timeLastPacketReceived > TPM2NET_PACKET_TIMEOUT) {
        // No data, nothing to do
        for(uint16_t i=0; i<strip.numPixels(); i++)
            strip.setPixelColor(i, 0);  // Turn Off all pixels
        strip.show();
        if(stop == TRUE) {return;}
    }
}
```

m-mcgowan, 2015-09-22:
What size packets are you sending? The code will fail if the packets are smaller than `EXPECTED_PACKED_SIZE`.  Since you're just copying the data to a buffer, you can simply use `receivePacket()`. 

Program defensively! Add a Serial.println() to output the received packet size. That will give you an indication about what's going on.

wmoecke, 2015-09-22:
The packets are all 199 bytes long. I've been monitoring the application
that sends the packets with Wireshark on my network, and they're all
consistent.

The sketch works when standalone but trying to integrate it with the code
I've referenced in my first post results in nothing being buffered to
parsePacket(); I have already debugged it and the issue is that
parsePacket() is always returning zero.

 "*C makes it easy to shoot yourself in the foot; C++ makes it harder, but
when you do it blows your whole leg off*" - *Bjarne Stroustrup*

wmoecke, 2015-09-22:
Tested replacing Udp.parsePacket() with Udp.receivePacket() in my code (integrated with the other code) and got the following compile error:
error: no matching function for call to 'UDP::receivePacket()'
I looked in the reference for UDP, there doesn't seem to be a reference to receivePacket().

steelydev, 2015-09-25:
@wmoecke Just for giggles and laughs, does the test code I first posted work on your Photon?

wmoecke, 2015-09-26:
Hi @steelydev, I haven't tested with your code.
Our codes are not too dissimilar from one another, but mine has the particular 'ability' to work perfectly, when it's the only code loaded in the photon. If I bundle this code with some other sketch (https://github.com/sparcules/Spark_Pixels/blob/master/Firmware/SparkPixels.ino), I never get any packets out of parsePacket(). Which is weird.

I will give your code a try, on both scenarios, and report back my results as soon as I can.
Thanks anyway.

wmoecke, 2015-09-26:
And btw, something I did not mention before: I have even tested placing the call to parsePacket() outside of the (bytes >= EXPECTED_PACKED_SIZE) condition evaluation, and still, when included in that other piece of code, zero packets.

steelydev, 2015-09-26:
@wmoecke You originally stated "when I tried adding my code into another existing piece of code" and the last post states: " If I bundle this code" 

How are you doing this?  Copy/paste?  .h and .cpp files?

wmoecke, 2015-09-28:
Simple answer. Imagine you'd simply add the loop() function as a function on that other code to be called the same way the existing ones are called. 
The necessary initialIzation and globals are transported as well. 
No separate headers or cpp. Just add it,, pure and simple.