particle-iot/device-os #470: Photon WiFi Password Length Limited

harrisonhjones, 2015-07-01:
See internal zendesk ticket #19752. The Photon seems to be limited to passwords < 23 characters.

m-mcgowan, 2015-09-13:
I don't routinely have access to Zendesk. It's a great help for me and others with the same issue if you could repeat the issue details. 

I did test with the information given here, and created a new test AP, with password of 43 characters.
I erased wifi credentials before each test, and managed to successfully connect to the AP using Serial configuration and SoftAP configuration. (I did this using a test program - I didn't use the mobile app, but the commands to the photon are the same.)

This shows that the Photon can handle passwords > 23 characters. 

I'm going to close this for now. Happy to re-open if you have more information that you'd like me to investigate.

closed 2015-09-13 by m-mcgowan

harrisonhjones, 2015-09-15:
This was back in July so I imagine the issue has been resolved.