particle-iot/device-os #471: Sprintf on develop branch - compiling error

LeoWW, 2015-07-01:
I can "make" (i.e. compile) my app on a previous version of the develop branch without errors. 
However, when pulling the latest develop branch, I get sprintf errors.

Issue is explained in more depth on Particle community page:
#1: http://community.particle.io/t/sprintf-not-working-on-develop/12525/2 
#2: http://community.particle.io/t/sprintf-errors-when-making-with-latest-firmware-develop-branch/12869/2

technobly, 2015-07-02:
I have definitely seen this as well, thanks for the head up!

m-mcgowan, 2015-07-13:
Here's some pointers that may help:

- be sure you are using version 2015 q2 of arm gcc. 
- Please try again since it's working for me in the latest develop branch.

technobly, 2015-07-13:
Definitely still an issue with the latest `develop` and 4.9.3 GCC.  @m-mcgowan remove all of your secret hidden bookshelf libraries! ;-)

```
`sprintf' referenced in section `.text.module_user_setup' of /var/folders/ts/bd82yb6s1ysdwfdtmvtgs0280000gn/T//ccp6A0VR.ltrans0.ltrans.o: defined in discarded section `.text' of rt_dynalib.o (symbol from plugin)
collect2: error: ld returned 1 exit status
make[1]: *** [../../../build/target/user-part/platform-6-m-lto/localip.elf] Error 1
make: *** [modules/photon/user-part] Error 2
```

m-mcgowan, 2015-07-13:
derp, my bad. I was sure it was fixed, now I recall I worked around the issue. Add COMPILE_LTO=n. I will make this the default. LTO is causing us more trouble than it's worth at present.

technobly, 2015-07-13:
`make clean all PLATFORM_ID=6 APP=localip COMPILE_LTO=n -s program-dfu`

with the following app, confirmed working:

```cpp
#include "application.h"

char myIpString[24];

void setup() {
    IPAddress myIp = WiFi.localIP();
    sprintf(myIpString, "%d.%d.%d.%d", myIp[0], myIp[1], myIp[2], myIp[3]);
    Spark.variable("ipPhoton", myIpString, STRING);
}

void loop() {
    
}
```

commit ed45a2c4 (2015-07-13): disable LTO since it causes linker errors on 2015 q2 ARM GCC 4.9.3. Fixes #471.
closed 2015-08-13 by m-mcgowan