particle-iot/device-os #472: Develop branch only: Wifi Listen and Credentials Reset modes don't change RGB LED to blue

nolanhergert, 2015-07-02:
Relatively minor bugs, but annoying for users that want to do wifi stuff. I am working exclusively with the develop branch, so I'm not sure about trunk.

My code is simply:

    SYSTEM_MODE(XXXXXX);
    WiFi.on();
    WiFi.connect();

When I get into [Start_Smart_Config](https://github.com/spark/firmware/blob/develop/system/src/system_network.cpp#L75) through various means (not having credentials on startup or holding the MODE button manually while running), the LED flashes the color previously set, instead of blue. 

**Note: Everything else (serial config of wifi) works great, yay!** 

Doing some debugging, it appears this is because LED_RGB_OVERRIDE is 1, because <unrecursing a few levels> [Spark_Signal](https://github.com/spark/firmware/blob/develop/system/src/system_cloud.cpp#L514) is getting called at some point. 

Below is a summary of the possible states. Incorrect means it blinks the previously set color, instead of blue.

State | AUTOMATIC | MANUAL
--------|------------------| -------------
Startup (no wifi creds) | CORRECT | Incorrect |
MODE Button held | Incorrect | Incorrect |

technobly, 2015-07-02:
@nolanhergert thanks for the very detailed report! Does it eventually flash blue after flashing the wrong color for a bit or does it flash the wrong color continuously?  Do you see any difference if you add Spark.process(); to your loop()?

nolanhergert, 2015-07-02:
Yes, the color is continuous. 

Ahh, I think I found the bug. I realize now that I set `RGB.control(true)` in my `loop()` code. When I don't do any RGB controlling in `loop()`, regardless of mode, it works as expected. 

What's weird is that even though I set `RGB.control(false)` later in my `loop()` code, the color is still wrong. It appears that holding MODE interrupts user code (it doesn't run at the end of `loop()`, which is fine) but keeps user state such as RGB control. That seems to me to be a bug still. 

w00t, I'm glad we are able to get closer on this! Sorry about the rabbit trail. :)

m-mcgowan, 2015-07-12:
The LED handling could do with a bit of a shakeup since it's not allowing the system to take control of the LED when it needs to. My initial thoughts on this are here for anyone interested, https://github.com/spark/firmware/issues/362#issuecomment-98295881

m-mcgowan, 2015-08-22:
Fixed!

closed 2015-08-22 by m-mcgowan

nolanhergert, 2015-08-24:
@m-mcgowan Thanks!