particle-iot/device-os #473: Debug build broken

marsus, 2015-07-02:
Building the develop branch with DEBUG_BUILD=y causes compilation errors:
```
src/photon/socket_hal.cpp:643:53: error: 'sd' was not declared in this scope
             DEBUG("Socket %d receive fail %d", (int)sd, int(result));
src/photon/socket_hal.cpp:657:56: error: 'sd' was not declared in this scope
         DEBUG("Socket %d receive bytes %d of %d", (int)sd, int(bytes_read), int(available));
```
The solution is to change `(int)sd` to `(int)tcp_socket->socket`; see diff below.

The build then fails with `system/src/main.cpp:199: undefined reference to 'log_print_'`, which is defined in `services/src/debug.c` but doesn't get linked in.

```
diff --git a/hal/src/photon/socket_hal.cpp b/hal/src/photon/socket_hal.cpp
index 3f5a397..b63058b 100644
--- a/hal/src/photon/socket_hal.cpp
+++ b/hal/src/photon/socket_hal.cpp
@@ -640,7 +640,7 @@ int read_packet_and_dispose(tcp_packet_t& packet, void* buff
         packet.offset = 0;
         wiced_result_t result = wiced_tcp_receive(tcp_socket, &packet.packet, _
         if (result!=WICED_SUCCESS && result!=WICED_TIMEOUT) {
-            DEBUG("Socket %d receive fail %d", (int)sd, int(result));
+            DEBUG("Socket %d receive fail %d", (int)tcp_socket->socket, int(res
             return -result;
         }
     }
@@ -654,7 +654,7 @@ int read_packet_and_dispose(tcp_packet_t& packet, void* buff
         memcpy(buffer, data, read);
         dispose = (total==read);
         bytes_read = read;
-        DEBUG("Socket %d receive bytes %d of %d", (int)sd, int(bytes_read), int
+        DEBUG("Socket %d receive bytes %d of %d", (int)tcp_socket->socket, int(
     }
     if (dispose) {
         packet.dispose_packet();
```

m-mcgowan, 2015-07-15:
Thanks for this. I'll add a debug build to our CI server so that these issues are caught much earlier!

m-mcgowan, 2015-07-16:
Hi @marsus - could you please pull the latest develop branch and verify debug build is working for you. Thanks!

closed 2015-08-11 by m-mcgowan