particle-iot/device-os #474: Start-up delay of Photon

bing1106, 2015-07-02:
There is a significantly longer start-up delay for the Photon of about 2.5 seconds. This is not desirable for applications that require quick response upon power-up. The delay was not an issue with the old Cores.

technobly, 2015-07-02:
I have seen this in manual mode, but more like 500ms.  Are you using MANUAL mode?  We've been discussing ways to fix this.  It revolves around needing to initialize the Wi-Fi, which can possibly be deferred until the user turns on Wi-Fi.

Same issue as:
https://github.com/spark/firmware-private/issues/63

bing1106, 2015-07-02:
Thanks for the quick reply! I have tried MANUAL mode and observed similar delay on the Photon as with SEMI_AUTOMATIC mode. The way I structure my code is that I would ideally run some code first in setup() then try to connect to cloud in SEMI_AUTOMATIC mode.

I am using the Cores in place of Photon for now and will wait for new DFU available that addresses this issue. Sooner or later I will start using the P1 modules as well. I understand this is work in progress.

btw I could not see the page you posted for some reason (404). Thanks!

technobly, 2015-07-02:
Yeah, that repo is private currently (and forever probably).  I just wanted to link the two issues.  Here's my first post from that issue:

---

I'm trying to build an application that needs to wake the Photon from deep sleep, and test if the WKP input is active still.  Typically you would debounce the input the same way you would if the system was up and running.  Then if the button press was just noise, the system could go right back to sleep.

I ran into a problem on the Photon with WKP from deep sleep, because it was taking:
**450-500ms before it would execute code in the setup() function**.  This runs code well after the user has let go of the button that woke the system up.  So basically if you wanted to debounce it, you'd need to press the button for about 600ms which just feels way too long to be natural.

Comparing this to the Core, it only takes:
**2ms to reach the setup() function.**
So in this case we can debounce the button for 30-50ms and things still feel nice and responsive from a UI perspective.

The goal should be, get to setup() in MANUAL mode as fast as possible.

Here's my test code:

```c++
#include "application.h"

uint32_t eventTime = 0;
uint32_t now = 0;

SYSTEM_MODE(MANUAL);

void setup() {
	now = millis();
	pinMode(17, INPUT); // WKP
	pinMode(D7, OUTPUT);

	if (digitalRead(17) == HIGH) {
    	
    	Spark.connect();
		while (!Spark.connected()) {
			Spark.process();
			digitalWrite(D7, !digitalRead(D7));
			delay(50);
		}
		
		Spark.publish("boot-time", String(now), 60, PRIVATE);

    	while (digitalRead(17) == HIGH) {
    		Spark.process(); // Since we are in manual mode, we need to run background tasks ourselves
    		digitalWrite(D7, !digitalRead(D7));
			delay(100);
		} // hang tight here until motion stops
    }
    else {
		// Go back to sleep
    	System.sleep(SLEEP_MODE_DEEP,1800);
    }
    eventTime = millis();
}

void loop() {
    if (millis() - eventTime < 10000UL) { // Stay away for 10 seconds
		digitalWrite(D7, !digitalRead(D7));
		delay(1000);
    }
    else {
    	// Go back to sleep
    	System.sleep(SLEEP_MODE_DEEP, 1800);
    }

    Spark.process(); // Since we are in manual mode, we need to run background tasks ourselves
}
```

technobly, 2015-08-10:
@bing1106 this might help you while you wait for an official fix to this startup delay.

You can do quite a bit in a Class Constructor which is executed very early on in the boot sequence:

```c++
class RunBeforeSetup {
public:
	RunBeforeSetup() {
		pinMode(D2, OUTPUT);
		digitalWrite(D2, HIGH);
		pinMode(D0, OUTPUT);
		analogWrite(D0, 127); // 50% duty cycle PWM
	}
};

RunBeforeSetup runBeforeSetup;

void setup() {
}

void loop() {
}
```

bing1106, 2015-08-10:
OMG, this is totally Brilliant!

I got it to light up the LED almost right when I plugged the P1 in. Now I will just need to incorporate this into my C++ and header files :)

Thank you so much!!

BracWebb, 2015-08-10:
I use this "hack" a lot -- it is great when needing to drive an external WDT :)

I have noticed that if my constructor has a lot of logic in it, I can actually get the device to lock up on bootup. Not sure what causes that and it has been awhile since I have tested, if it was just related to early stability issues on the Photon.

bing1106, 2015-08-10:
I am using the P1 right now and it works well for my system initialization. The fix is great for now.

In my constructor, I basically have the two following actions:
- initialize I2C,
- calculate some values and send to I2C device

I don't think there is much logic in it so I am not getting and lock ups. I have to say that I find the P1s to be more stable than the Photons.

m-mcgowan, 2015-08-22:
In 0.4.4, we also added the `STARTUP()` macro that allows you to conveniently execute code early in the startup process.  For example, define a function `void init_my_stuff()` and then you can write
```
STARTUP(init_my_stuff()); 
```
in global scope, and your code will be executed early.

m-mcgowan, 2015-08-22:
This looks resolved to me so going to close. Feel free to re-open if there is more that needs to be discussed.

closed 2015-08-22 by m-mcgowan

bing1106, 2015-08-22:
Thanks so much!!!!

technobly, 2016-06-20:
FYI: I used my test app above on a Photon running 0.5.1 and the boot time reported was 44ms.  Not bad for not having to use the STARTUP() macro.