particle-iot/device-os #475: Move `BUFFER_LENGTH` to `i2c_hal.h` header

loopj, 2015-07-03:
With the new HAL refactor, the `BUFFER_LENGTH` define has been moved from a header into `i2c_hal.c`, meaning it is no longer accessible:
https://github.com/spark/firmware/blob/develop/hal/src/core/i2c_hal.c#L40

To remain Arduino-compatible, I'd like to recommend this `#define` be moved into `i2c_hal.h`, since that will ensure it can be used by client applications which care about the i2c buffer length (eg. i2cdevlib).

technobly, 2015-07-06:
Hey @loopj, do you just need to know that the buffers are 32 bytes, or do you want to be able to modify them as well?  Taking a quick look at the Arduino reference, I don't see mention of BUFFER_LENGTH.  It might be nice to add an API that can modify this buffer size, and request what the current buffer size is.  You also won't have to know to include i2c_hal.h and that BUFFER_LENGTH exists if we add an API for these.

loopj, 2015-07-06:
Hey @technobly, it is unlikely that I'll need to modify this.

Here's where `BUFFER_LENGTH` is defined in `Wire.h`:
https://github.com/codebendercc/arduino-library-files/blob/master/libraries/Wire/Wire.h#L28

Here's where `BUFFER_LENGTH` is used in the popular `i2cdevlib` library:
https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/I2Cdev/I2Cdev.cpp#L249

Adding `BUFFER_LENGTH` to `i2c_hal.h` would allow drop-in compatibility, but you could also argue that `i2cdevlib` should define its own buffer here.

technobly, 2015-07-07:
Ok cool, that makes sense @loopj.  One problem I have with simply exposing BUFFER_LENGTH is it's super non-descriptive.  Sure it's unique to the I2C library, but it could be one of any number of buffers created on the system.  If we exposed it in the header file I would also want a better user-facing API to interact with it.  Something we could document and be proud of :)  Without that I'd be inclined to change BUFFER_LENGTH to something like I2C_BUFFER_LENGTH, but that would break your use case. I suppose these could be two separate items though, so if you want to submit a PR for the change of moving BUFFER_LENGTH to i2c_hal.h that would be helpful.

m-mcgowan, 2015-08-11:
This and some of the other non-conscientous design choices in the old Arduino libraries we're deliberately not supporting since they would create trouble for us.

I agree with @technobly that naming this I2C_BUFFER_LENGTH would be a good start to prevent putting generic names in the global namespace. Then the i2c library can add a short bit of code like this to the top of the file:

```
#ifdef PARTICLE
// specific changes for particle
#define BUFFER_LENGTH I2C_BUFFER_LENGTH
#endif
```

Then the library code (those parts that refer to BUFFER_LENGTH) will compile without changes.

How does that sound?

loopj, 2015-08-11:
This sounds like a perfect compromise, I completely agree that having
genericly named defines is a bad idea.

commit 4b09814e (2015-08-11): Added I2C_BUFFER_LENGTH in i2c_hal.h for partial arduino compatibility. Fixes #475.
commit 95998a12 (2015-08-26): Added I2C_BUFFER_LENGTH in i2c_hal.h for partial arduino compatibility. Fixes #475.
closed 2015-08-13 by m-mcgowan