particle-iot/device-os #476: [Photon] Bluetooth signal pins

kennethlimcp, 2015-07-05:
In the docs:

```
When these pads are programmed to be used as a Bluetooth coexistence interface, they're set as high impedance on power up and reset. Alternatively, they can be individually programmed to be used as GPIOs through software control. They can also be programmed to have an internal pull-up or pull-down resistor.
```

Is there more information available as to whether this has been tested and how it can be done?

technobly, 2015-07-06:
This is a feature supported by the broadcom radio and WICED API, not an inherent feature within the STM32F205.  We haven't tested this or developed any solution with it yet, just exposed the pins on the Photon so that it's possible to use.  If you find more information, like a Broadcom appnote describing how to use this feature... that would be helpful! ;-)

kennethlimcp, 2015-07-06:
So far what i see is that controlling the pin is not possible but might need to look into the latest WICED SDK. As far as i checked, the pins are not controlled anywhere or even defined by `platform.h`

m-mcgowan, 2015-09-14:
Any resolution to this? Kenneth, what kind of description would you prefer in the docs?

kennethlimcp, 2015-09-14:
We need to removethe descrption in the documentation that those pins can be used as GPIO

kennethlimcp, 2015-09-16:
@m-mcgowan, i have updated the docs here: https://github.com/spark/docs/commit/a8cabd7c2607a3d24e0e27efd49e0c18c88b9ae9.

Let me know if this can be closed. Feel free!

technobly, 2015-09-16:
@kennethlimcp I'd rather not push changes to the datasheets directly.  I try to stage the changes in the docs issue # 19 and then make a revision to the datasheets as a collection of changes with a revision.

I'd like to still think we can control these pins based on the USI datasheet:
>Multiplexed BT_Coex pins. When programmed are high impedance on power up and reset. Subsequently, they can be individually programmed to become inputs or outputs through software control. They can also be programmed to have internal pull-up or pull-down resistor.

I'll ask USI what the interface is to control these pins and if for sure they can be used as GPIO.

kennethlimcp, 2015-09-16:
@technobly, ack for the datasheet revision updating flow. I have reverted the PR: https://github.com/spark/docs/commit/b370cdde9f2865c387912c101536c877adc5b720 and reopened the issue. Thanks!

ScruffR, 2015-09-17:
One problem with adding info from the datasheet "unreflected" and without the required additional info in place might turn out like this
```
We've already turned boards, based on the datasheet and are in the process of validating the functionality. If we can't use these IO pins we'll have to look at adding an IO expander and re-spinning the boards.
```
*(from https://community.particle.io/t/external-coexistence-interface-pin-reference/15612/3)*

So I'm with Kenneth. It should not have been stated in the first place or with an explicit mentioning ***"Not supported yet!!***

dseemakurty, 2016-02-24:
Can the ble coexistence interface pins be used for... coexistence functions? Any idea of how they work? 

Does the BTCX_TXCONF pin automatically go high when wifi is not operating? Does driving the BTCX_RF_ACTIVE pin high or low prevent the Photon/P1 from transmitting Wifi? 

Are these pins currently functional for anything or are they just broken out and currently unsupported?

technobly, 2016-03-17:
@dseemakurty the BT COEX pins are broken out and currently unsupported.  Adding support for them can be bumped up the priority list however we have not had much request for it.  Most people want to use the pins as GPIO, which is also currently not practical.

Bacichetti, 2016-12-06:
@technobly , my new board uses Bluetooth and coexistence pins will be necessary for controlling the radios (to simplify or make certification possible). It doesn't look to work, could you test it and update the documentation. If you don't plan doing that, is there another way to take control over when the Broadcom chip make radio transmissionsavailable?

@dseemakurty, have you managed to get it working?

There are a few more requests for the coexistence over the Community, maybe user are not certifying their products OR just chosen other option on the market where this option is available. 

I need to say, I understand that's a feature with very low priority to most user using Bluetooth, but few "commercial" user like me, hopefully with thousands of sells in the next year :) , have this feature at the top of our lists.

technobly, 2016-12-10:
Re: BT coex functionality, I'll add this to my list of things to look into over the holiday break ;-)

Bacichetti, 2016-12-13:
Thanks @technobly, I really appreciate!

Bacichetti, 2018-05-30:
Hi @technobly, any news on this feature??

closed 2018-12-24 by kennethlimcp