particle-iot/device-os #477: Spark Core breathing white continuously (using v0.3.4 core-firmware)

javier-rodas, 2015-07-08:
Hello,

I haven't find any duplicated issue in github so I have created a new one. Let me know if this is solved in any other issue...

---

I have been testing 5 Spark Cores for 3 months and one of them suddenly started **breathing white** continuously, not flashing green or blue anymore, and so on.

It looks like it continuously resets at boot time and doesn't continue to the following steps, of activating the WiFi, etc.... or maybe its because of a hardware failure, I don't know.

The Spark Core has a modified versión of the core-firmware v0.3.4 installed, with a custom user program on top that is running without problem in the rest of 4 cores that we have.

I'm able to put it in dfu more by pressing the 2 buttons, etc. and backup its keys by USB, etc. but after reseting it again, it starts breathing white again continuously.

Before trying to re-flash its firmware again, I would like to know:
- Is there any way to debug what is actually wrong in the module? Maybe by USB or using an ST-LINK/V2?
- What are the recommended steps to replace its firmware without having to register the keys again in the Spark Cloud? I'm using at present a local Spark Cloud server with an specific public key, static IP, etc.
- Which firmware versión do you recommend for the **Spark Core** at present, the 0.3.4 core-firmware or the "develop" branch versión with the HAL layer, etc.?  In the case of using the develop branch versión, is it 100% compatible with the local Spark Cloud server that I'm currently using? or I have to tune it in any way to make them work together?

kennethlimcp, 2015-07-08:
A breathing white core indicates that the CC3000 might not be responding to commands so that might be a hardware issue.

javier-rodas, 2015-07-08:
@kennethlimcp  I have reflashed the tinker app by USB and it continues flashing white, so I guess it's definitely a hardware issue.

Could you please tell me what is the normal procedure to send a failure Core back to you?
The module is still in warranty and was bought though your official store webpage, but I would like to know if it's possible to send it back to any SAT place in Europe instead of sending it to the US.

javier-rodas, 2015-07-09:
I guess I will contact the support people... The rest is solved.

closed 2015-07-09 by javier-rodas