particle-iot/device-os #478: Not able to configure a Photon in Soft-AP mode from an iPhone

javier-rodas, 2015-07-08:
I correctly setup by USB a brand new Photon I've received with the WiFi credentials and so on. First I tried by WiFi using an iPhone without success, so I finally did it by cable.

After the first connection to the Particle Cloud the firmware was upgraded automatically (LED flashing magenta) and after rebooting correctly I could claim the Photon from the build website.

The main issue I have observed is that I could not configure the Photon via WiFi using an iPhone, not with the Photon stock pre-flashed firmware nor after this first firmware upgrade performed today (8 July, 2015), few minutes ago.

I have tried to connect with and **iPhone5**, with **iOS 8.1** and with the **latest iOS 8.4**. In both cases when I enter in the Settings > WiFi configuration panel and tried to connect to the Photon-XXXX soft-ap, after <1s it automatically rejects to connect (and the "Photon-XXXX" comes back to the available AP list, without showing any error message). Some times if I insist it finally shows an error message popup, that it cannot connect to that network.
I have tried restarting the WiFi, rebooting the iPhone, etc. No way!!.

---

Afterwards, I've tried connecting from my Macbook and it worked!!. I've gotten a 192.168.0.2 dynamic IP by DHCP.

---

So, at this point I would like to know:
- Is there a workaround to debug this connection problem from my iPhone and be able to connect to the Photon in Soft-AP mode?.
- Is there a way/tool/website API to configure/communicate with the Photons in Soft-AP mode, without using an iPhone or an Android smartphone?. What is its current availability state?

m-mcgowan, 2015-07-08:
cc: @idokleinman

javier-rodas, 2015-08-06:
Hello @m-mcgowan cc: @idokleinman 

I have discovered that the problem appears only in a really 2.4GHz WiFi crowded environment, with more than 15 (or even more WiFis available). In this environment it is quite impossible to connect to the Photon in SoftAP mode from any iOS or Android device. I've tried with 4 different smartphones/tablets, with different OSs.

However, when I moved far away from the "problematic" building into my car, I was instantaneously able to connect to the Photon with all the four devices nearly instantaneously!!. So, issue solved.

closed 2015-08-06 by javier-rodas

rpruizc, 2015-08-06:
Really? Issue solved/closed? We experienced the same issue with nowhere near to 15 (although there certainly were more than a couple devices in the room) right in the middle of a very important demo. IMO, this issue should be left open for further investigation.

kennethlimcp, 2015-08-06:
@rpruiz you can open a new issue and reference this one.