particle-iot/device-os #480: System.sleep(5) not turning Wi-Fi back on after 5 seconds.

technobly, 2015-07-08:
System.sleep(5) not turning Wi-Fi back on after 5 seconds.  System.sleep is supposed to turn of Wi-Fi and continue running code, which it does.  However after 5 seconds Wi-Fi is not turned back on and reconnected automatically, which is not stated anywhere but implied with `sleep(5)`.
https://community.particle.io/t/not-sure-how-to-use-sleep-mode/13082/4

I have created an app to demonstrate this, if you read the comments you'll see that the D7 LED is toggling every second, which makes it easy to count to 20, after which the RGB should be WHITE.  5 seconds later it should be blinking green but does not.  then at the 30 second mark we force a reconnect.  At the 40 second mark the system will try to sleep(5) again, and this process will repeat over and over.

## Test App:
```cpp
unsigned long lastTOGGLE;
unsigned long lastSLEEP;
#define now() millis()
#define elapsedLED() (now()-lastTOGGLE)
#define elapsedSLEEP() (now()-lastSLEEP)
#define toggleLED() (digitalWrite(D7, !digitalRead(D7)))

void setup(){
    pinMode(D7,OUTPUT);
    lastTOGGLE = now();
    lastSLEEP = now();
}

void loop(){

    // Toggle LED every 1 Second
    if( elapsedLED() >= 1000UL ) {
        lastTOGGLE = now();
        toggleLED();
    }
    
    // Sleep (turn off Wi-Fi) after every 20 seconds of runtime
    if( elapsedSLEEP() >= 20000UL ) {
        System.sleep(5);
        lastSLEEP = now();
    }
    
    /*
     * IT IS EXPECTED HERE AFTER 5 MORE SECONDS OF 
     * RUNTIME THAT WE WILL WAKE FROM SLEEP AND
     * TURN WI-FI BACK ON, BUT IT DOESN'T HAPPEN
     */
    
    // After 5 more seconds of runtime past when we 
    // we,re supposed to reconnect, force a reconnect.
    if( now()-lastSLEEP > 10000UL ) {
        if (!WiFi.ready()) {
            WiFi.connect();
            while (WiFi.connecting())
            {
                toggleLED();
                delay(100);
                Spark.process();
            }
        }
    }
}
```

technobly, 2015-07-08:
Tried this on a Core and had to modify the above example.

It does work on Core, but even though the WiFi comes back on it does not reconnect to the Cloud automatically (thus some different code to force WiFi and Cloud reconnect, i.e. Spark.connect()). I tried using Spark.connect() in the case of the Photon as well and that did not force it to connect to Wi-Fi first.

We should discuss the expected behavior and then implement a solution.  It's possible that the Core's behavior is what we want, although in AUTOMATIC mode I would assume if you reconnected WiFi (aka waking back up from Spark.sleep(5)) that it would also automatically reconnect to the Cloud (and it does not).

m-mcgowan, 2015-07-08:
Agreed. I think in AUTOMATIC mode wakeup from sleep connects to the cloud. I believe that's the intended behavior, but it may have regressed.  This is a great candidate problem for starting an integration testing framework.

m-mcgowan, 2015-07-16:
The RTC alarm handler only turns on wifi for automatic mode. https://github.com/spark/firmware/blob/develop/system/src/main.cpp#L163-L170

I'm guessing this has been done deliberately, but it's not clear why. I would remove the conditional and enable WiFi for all modes.

@satishgn Any thoughts about this?

m-mcgowan, 2015-08-22:
I made this to save the current wifi and cloud connection state and to restore these on wakeup. To my mind this seems the most natural  course of action to resume the state prior to sleeping. This means the cloud is reconnected even in manual mode if it was connected at the time `System.sleep(for_how_long)` was called.

m-mcgowan, 2015-08-23:
@technobly - I'll leave it up to you if you want to confirm the fix or just close.

technobly, 2015-08-24:
@m-mcgowan can you post whatever code you used to test this to make sure if I or anyone else tests it that we try to cover more bases?  Thanks!

m-mcgowan, 2015-08-24:
Here's the code - a variation of the manual connection tester.

Use 'C', 'W', 'c', 'w', 'o', 'O', to set the connection state, then hit 's' to sleep.

```
#include "application.h"

SYSTEM_MODE(SEMI_AUTOMATIC);

void tcp_connect();

TCPClient client;

void setup()
{
  // Make sure your Serial Terminal app is closed before powering your Core
  Serial.begin(9600);

  // Now open your Serial Terminal, and hit any key to continue!
  while(!Serial.available()) Spark.process();

}

void loop()
{
  if (Serial.available()) {
    int c = Serial.read();
    switch (c) {
      case 'C': Spark.connect(); break;
      case 'c': Spark.disconnect(); break;
      case 'W': WiFi.connect(); break;
      case 'w': WiFi.disconnect(); break;
      case 'T': tcp_connect(); break;
      case 't': client.stop(); break;
      case 'O': WiFi.on(); break;
      case 'o': WiFi.off(); break;
      case 'L': WiFi.listen(); break;
        case 'x':WiFi.clearCredentials(); break;
        case 'l': wlan_smart_config_finalize();
    case 's' : System.sleep(5);
    }
  }

  if (client.available())
  {
    char c = client.read();
    Serial.print(c);
  }
}

void tcp_connect() {
  Serial.println("connecting...");

  if (client.connect("www.textfiles.com", 80))
  {
    Serial.println("connected");
    client.println("GET /100/ad.txt HTTP/1.0");
    client.println("Host: www.textfiles.com");
    client.println("Content-Length: 0");
    client.println();
  }
  else
  {
    Serial.println("connection failed");
  }
}```

bprobbins, 2015-08-24:
Please see #580

commit d182e331 (2015-08-22): this was not working as a result of flags. Before sleep, network_off() is called, which doesn't set the deliberate disconnect flag. In taking down wifi, HAL_WLAN_notify_disconnected is called, which then sets up a network watchdog for 2 seconds. When the 5 second timeout causes wifi to wake up, the watchdog has tripped, and this causes the exceptional path through manage_network() (same as when wifi is reset.)  The combination of flags meant manage_network() did nothing and never restarted wifi on the background loop.

The fix is to ensure WLAN_DISCONNECT is set when sleeping (by calling netwo [... cut]
commit b4817c1e (2015-08-25): Updated HAL_RTC_Configuration() - Fixes #480, #580
closed 2015-08-26 by m-mcgowan