particle-iot/device-os #482: `particle serial ...` commands often fail in SYSTEM_MODE(MANUAL) on Core

pomplesiegel, 2015-07-13:
Our firmware (running in SYSTEM_MODE(MANUAL), when put in listening mode, often times out when running particle-cli commands like `particle serial identify` and `particle setup wifi`. Sometimes the first couple tries will work, but then all subsequent calls will fail. 

When tinker is loaded, the problem totally disappears. It's possible that this is related to SYSTEM_MODE(MANUAL), but I am unsure.

Below is a very simple program replicating the basic setup of our firmware. **This reproduces the issue.** After placing the core in listening mode, the first call passed. All subsequent calls failed, even following a reset of both the Core and computer.

```cpp
SYSTEM_MODE(MANUAL);

void setup()
{

  delay(3000);
  Serial.begin(9600); //Begin USB serial session
  WiFi.off(); 
  
  Wire.setSpeed(CLOCK_SPEED_100KHZ);
  Wire.stretchClock(true); //enable clock stretching
  Wire.begin();

  WiFi.on(); //Turn on Wifi chipset
  Spark.connect();
  
  Serial.println("Finished setup()");
}

void loop()
{
  Spark.process()
  delay(10);
}
```

Originally entered as a [particle-cli issue.] (https://github.com/spark/particle-cli/issues/56#issuecomment-120685583) Additional background info there.

m-mcgowan, 2015-07-13:
Thanks for reporting this!  Are you able to build using the local toolchain? If so, I'm interested to know if switching to the develop branch and building firmware there fixes the problem. (The 0.3.4 release in the master branch is from October last year - we have made many fixes since then.)

pomplesiegel, 2015-07-13:
@m-mcgowan, I just set up my local build. Good to have a reason to do so. Thanks!

Checked out `develop` and built w/ the code above replacing application.cpp. 

Good news: The timeouts when calling `particle serial identify` and `particle setup wifi` are gone! 

Weird news: At startup my Core is now just breathing blue, without successfully connecting to the cloud. The only way I can get it breathing cyan is to place it into listening mode and successfully calling `particle setup wifi`. 

I've tried
* Adding Hard-coded wifi creds in setup()
* Adding ```if(!Spark.connected())
    Spark.connect();``` to loop()
* Removing the 10ms delay in loop() 
* Remvoing WiFi.off() setup()

but to no avail. Even after getting it to breathing cyan, if I reset the spark, it goes back to breathing blue. 

Any thoughts?

pomplesiegel, 2015-08-31:
Seeing the same behavior on a Photon running 0.4.4 or a Core running FW from the master build.

Basically,

Many particle-cli commands time out with a error similar to

````sh
Something went wrong Serial prompt timed out - Please try restarting your device
Serial error: { '0': [Error: Serialport not open.] }
````

if a Photon or Core is running in SYSTEM_MODE(MANUAL)

These commands include
`particle setup`
`particle setup wifi`
`particle serial wifi`
`particle serial identify`

m-mcgowan, 2015-09-01:
The device should be placed in listening mode prior to invoking these commands.  From the particle-cli [docs](https://docs.particle.io/reference/cli/)

> particle setup wifi
Helpful shortcut for adding another wifi network to a device connected over USB. Make sure your device is connected via a USB cable, and is slow blinking blue listening mode.

I hope that helps.

pomplesiegel, 2015-09-01:
Hi @m-mcgowan, Just to be clear, the device is in listening mode (blinking blue) when these commands are called. 

The easiest way to reproduce this is with the simple program

````cpp
SYSTEM_MODE(MANUAL);

void setup()
{
  Serial.begin(9600); //Begin USB serial session
  WiFi.off(); 

  WiFi.on()
  WiFi.connect()
  Particle.connect();

  Serial.println("Finished setup()");
}

void loop()
{
  Particle.process()
  delay(10);
}
````

Once the device has reached steady state, it may be placed in listening mode, at which time these calls time out.

m-mcgowan, 2015-09-01:
Thanks for clarifying! I'll try out the test case to see if I can reproduce the issue.

m-mcgowan, 2015-09-02:
I just tried this and cannot reproduce the issue on OS X 10.10.5. Which OS are you using? If Windows, please be sure that all programs using the serial port are closed, including previous instances of particle-cli.  Windows has very "unfriendly" behavior regarding serial.

pomplesiegel, 2015-09-02:
Interesting. I'm running OS X 10.10.4 w/ the latest particle-cli, no serial ports open 

The 0.4.4 system FW was flashed onto the board from your precompiled ".bin"s, but the code shown above was compiled remotely using particle-cli. Is that how you're building as well?

pomplesiegel, 2015-09-02:
False alarm! Sorry for the hassle. 
Perhaps the original `0.4.4` flash to this Photon failed, as now after running `particle update` everything seems OK.

closed 2015-09-02 by pomplesiegel