particle-iot/device-os #483: Tone on P1 and core do not match

wesner0019, 2015-07-14:
It looks like the P1 has PWM on the below pins:
"From the datasheet online
PWM is available on D0, D1, D2, D3, A4, A5, WKP, RX, TX with a caveat:
PWM timer peripheral is duplicated on two pins (A5/D2) and (A4/D3) for 7
total independent PWM outputs. For example: PWM may be used on A5 while
D2 is used as a GPIO, or D2 as a PWM while A5 is used as an analog
input. However A5 and D2 cannot be used as independently controlled PWM
outputs at the same time."

But the below pins are defined for tone
"tone()
Generates a square wave of the specified frequency and duration (and
50% duty cycle) on a timer channel pin (D0, D1, A0, A1, A4, A5, A6, A7,
RX, TX). "

With this current setup tone will only work the P1 with PWM pins (D0, D1, A4, A5, WKP,
RX, TX)?

If the firmware was updated then pins D2, D3 for the P1 could also have the Tone function?

m-mcgowan, 2015-07-14:
@technobly, @satishgn  - any thoughts on this?  Would it be worth making a chart showing pins/timers and capabilities?

technobly, 2015-07-14:
The section on tone() in the Docs just does not have the correct info for the Photon.  See the section on PWM which shows the correct info for Core/Photon PWM capability.  http://docs.particle.io/photon/firmware/#i-o-analogwrite Any pin that supports PWM will also support tone().  I'll add an issue to the Docs repo.  Thanks @wesner0019 

And yes, you could dynamically add and remove functionality from one pin A5 and add it to D2, or vice versa.  Same with A4/D3.  So you could use tone() on A4, and then call noTone(A4) to remove it, then call tone() on D3, then call noTone(D3) to remove it.

wesner0019, 2015-07-14:
@m-mcgowan, do you think the tone function could be added to the D2 and D3 pins for the P1?

technobly, 2015-07-14:
@wesner0019 tone() already exists afaik on D2 and D3 pins.  Have you tried this?  They just cannot be used at the same time as A4/A5 with independent tone values.  For example you can't apply a 500Hz tone to D3 and 1000Hz tone to A4 at the same time, because both of those pins share the same hardware TIMER in the STM32.  Same goes for the D2/A5 pair.

wesner0019, 2015-07-14:
@technobly, i tried tone(D3,2150) on ONLY D3 but it does not work. The P1 has PWM on D3 but not the tone function according to the documentation for tone(), the tone function is not available on D3 unless I'm misunderstanding how the tone function works. 

From the documentation on tone (D3 is not mentioned):
"Generates a square wave of the specified frequency and duration (and
50% duty cycle) on a timer channel pin (D0, D1, A0, A1, A4, A5, A6, A7,
RX, TX). "

wesner0019, 2015-07-17:
@technobly, just seeing if anyone confirmed that tone either does or does not work on pins D2 and D3?

technobly, 2015-07-17:
Just tested with this app:
```
// The Photon should have 9 PWM pins: D0, D1, D2, D3, A4, A5, A7, RX and TX.
//
// EXAMPLE USAGE
// Plays a melody - Connect small speaker to speakerPin
int speakerPin = D0;

// notes in the melody:
int melody[] = {1908,2551,2551,2273,2551,0,2024,1908}; //C4,G3,G3,A3,G3,0,B3,C4

// note durations: 4 = quarter note, 8 = eighth note, etc.:
int noteDurations[] = {4,8,8,4,4,4,4,4 };

void setup() {
  // iterate over the notes of the melody:
  for (int thisNote = 0; thisNote < 8; thisNote++) {

    // to calculate the note duration, take one second
    // divided by the note type.
    //e.g. quarter note = 1000 / 4, eighth note = 1000/8, etc.
    int noteDuration = 1000/noteDurations[thisNote];
    tone(speakerPin, melody[thisNote],noteDuration);

    // to distinguish the notes, set a minimum time between them.
    // the note's duration + 30% seems to work well:
    int pauseBetweenNotes = noteDuration * 1.30;
    delay(pauseBetweenNotes);
    // stop the tone playing:
    noTone(speakerPin);
  }
}
```
Confirmed that D2, and D3 do not work as expected (they do seem to somewhat respond at the right timing... just not outputting a tone/PWM).

Also found RX and TX do not do anything at all when tone() is called.

wesner0019, 2015-07-17:
Thanks for looking into this. Do you think this will be a simple fix?

wesner0019, 2015-07-17:
I noticed that firmware/hal/SRC/photon does not have a tone_hal.c in it while the firmware/hal/SRC/core does. Not sure if this means anything.

technobly, 2015-07-19:
I was looking into this the other night, but my attempts to add the extra pins quickly didn't work as I planned.  Thought I would share what I did just in case it could help unblock this issue.  It appears with and without the following code, you need to add A5 & D2 and call tone() for both of them at the same time to get tone() on D2.  That's really bizarre and I'm sure that's not ok with you, or you'd probably have just used tone() on A5.  The tone_hal.c is in the hal/src/stm32f2xx folder.  Need to look into why A5 is some kind of enabler for D2, maybe this is because of Alternate Pin Functions.  See the following modifications for what I was attempting to do.  I had attempted to use `.timer_ccr != 0` to test if the pin had been set up for a tone or not, probably going to need a more explicit variable for this.

```cpp
static void Tone_TIM3_Handler(void)
{
    uint16_t capture;
    STM32_Pin_Info* PIN_MAP = HAL_Pin_Map();
    
    // A4 & D3
    if (TIM_GetITStatus(TIM3, TIM_IT_CC1) != RESET)
    {
        TIM_ClearITPendingBit(TIM3, TIM_IT_CC1 );
        capture = TIM_GetCapture1(TIM3);
        
        // No need to update TIM_SetCompare2 twice if both pins used
        if (PIN_MAP[14].timer_ccr != 0) {
            TIM_SetCompare1(TIM3, capture + PIN_MAP[14].timer_ccr);
        }
        else if (PIN_MAP[3].timer_ccr != 0) {
            TIM_SetCompare1(TIM3, capture + PIN_MAP[3].timer_ccr);
        }

        // Update both if both pins are used
        if (PIN_MAP[14].timer_ccr != 0) {
            if(PIN_MAP[14].user_property != -1)
            {
                if (PIN_MAP[14].user_property > 0)
                {
                    PIN_MAP[14].user_property -= 1;
                }
                else
                {
                    HAL_Tone_Stop(14);
                }
            }
        }
        if (PIN_MAP[3].timer_ccr != 0) {
            if(PIN_MAP[3].user_property != -1)
            {
                if (PIN_MAP[3].user_property > 0)
                {
                    PIN_MAP[3].user_property -= 1;
                }
                else
                {
                    HAL_Tone_Stop(3);
                }
            }
        }
    }

    // A5 & D2
    if (TIM_GetITStatus(TIM3, TIM_IT_CC2) != RESET)
    {
        TIM_ClearITPendingBit(TIM3, TIM_IT_CC2);
        capture = TIM_GetCapture2(TIM3);

        // No need to update TIM_SetCompare2 twice if both pins used
        if (PIN_MAP[15].timer_ccr != 0) {
            TIM_SetCompare2(TIM3, capture + PIN_MAP[15].timer_ccr);
        }
        else 
        if (PIN_MAP[2].timer_ccr != 0)
        {
            TIM_SetCompare2(TIM3, capture + PIN_MAP[2].timer_ccr);
        }

        // Update both if both pins are used
        if (PIN_MAP[15].timer_ccr != 0) {
            if(PIN_MAP[15].user_property != -1)
            {
                if (PIN_MAP[15].user_property > 0)
                {
                    PIN_MAP[15].user_property -= 1;
                }
                else
                {
                    HAL_Tone_Stop(15);
                }
            }
        }
        if (PIN_MAP[2].timer_ccr != 0) {
            if(PIN_MAP[2].user_property != -1)
            {
                if (PIN_MAP[2].user_property > 0)
                {
                    PIN_MAP[2].user_property -= 1;
                }
                else
                {
                    HAL_Tone_Stop(2);
                }
            }
        }
    }
}
```

## and my user app:

```cpp
#include "application.h"
// The Photon should have 9 tone pins: D0, D1, D2, D3, A4, A5, A7, RX and TX.
//
// EXAMPLE USAGE
// Plays a melody - Connect small speaker to speakerPin
int speakerPin1 = D2;
int speakerPin2 = A5;

#define NOTE_G3   2551
#define NOTE_C5   956
#define NOTE_E5   759
#define NOTE_G5   638

// notes in the melody:
int melody[] = {NOTE_E5,NOTE_E5,NOTE_E5,0,NOTE_C5,NOTE_E5,0,NOTE_G5,0,NOTE_G3};
// note durations: 4 = quarter note, 8 = eighth note, etc.:
int noteDurations[] = {8,8,8,4,8,8,8,8,8,8};

void setup() {
  // iterate over the notes of the melody:
  for (int thisNote = 0; thisNote < 8; thisNote++) {

    // to calculate the note duration, take one second
    // divided by the note type.
    //e.g. quarter note = 1000 / 4, eighth note = 1000/8, etc.
    int noteDuration = 1000/noteDurations[thisNote];
    tone(speakerPin1, (melody[thisNote]!=0)?(500000/melody[thisNote]):0,noteDuration);
    tone(speakerPin2, (melody[thisNote]!=0)?(500000/melody[thisNote]):0,noteDuration);

    // to distinguish the notes, set a minimum time between them.
    // the note's duration + 30% seems to work well:
    int pauseBetweenNotes = noteDuration * 1.30;
    delay(pauseBetweenNotes);
    // stop the tone playing:
    noTone(speakerPin1);
    noTone(speakerPin2);
  }
}
```

wesner0019, 2015-07-19:
@technobly, thanks for looking into this. Do you think the RX and TX pins can use tone? I can swap with D3 to RX or TX.

Thanks,

technobly, 2015-07-19:
Keep in mind I didn't spend much time on this, as most of my time is devoted to the Electron currently.  That said I am less certain about how to fix RX and TX as they seem to be coded properly but do not even glitch when used.  One difference to note is TIM1_CC_IRQn() which is a specific capture compare interrupt vs TIMx_IRQn() of the others which is a global timer interrupt.  There may be some tricks to using it that are explained in the reference manual.

wesner0019, 2015-07-20:
@technobly , I can understand that, I assume the tone function is not top priority as of now. 
For now, I will take one of my features from A7 and place it on a spare pin in hopes future updates will allow ADC on some of the spares and place my buzzer on A7. 

Thanks for being forward with the priories of issues being solved. I don't have the experiences to solve this, so for the time being, I can just move on by making the above change.

satishgn, 2015-07-23:
Investigating #483

satishgn, 2015-07-24:
Tone should work on D2,D3 and RX,TX after merging the branch feature/photon-pwm-and-tone-fix commits

wesner0019, 2015-07-25:
@satishgn, just wanted to give you a heads up to maybe add something to documentation about the use of tone on D2 or D3 while using the SPI interface. I have a SPI TFT screen connected to my P1 and just tested using D3 with a tone function. After calling tone on D3 the SPI screen is flickering rapidly even after I call noTone.

technobly, 2015-07-25:
Thanks for the heads up @wesner0019.  I think I'd much rather there not be any glitching on A4/A5 when D2/D3 is in use by tone() and A4/A5 are in use by SPI.  @satishgn would you please take a look at this and update this issue afterwards?  I'll re-open the issue so we don't forget.

wesner0019, 2015-07-27:
@satishgn @technobly, I just wanted to add that on my tft I'm controlling brightness of the back light through PWM using a N-MOSFT connected to D2. Maybe D2 is also affected in some way? Not sure if this helps.

technobly, 2015-07-27:
It helps to know you are using D2 as a PWM and not tone(), which this issue is about.  Is that right @wesner0019 ?  I would suspect both would have an issue potentially though.

wesner0019, 2015-07-27:
@technobly , I'm using both pins and both functions.

D2 is PWM works fine until tone is called on D3 or there is an issue with SPI on A4, but the screen is flickering not sure which one would cause flickering
D3 is Tone and this functions as it should.

satishgn, 2015-07-27:
@wesner0019, I don't think the screen flickering as an issue with SPI but because of the backlight PWM since D2 and D3 share the same timer peripheral viz. TIM3. When in PWM mode, D2 and D3 should output (fixed autoreload) 500Hz period with (variable ccr) duty cycle and the TIM3 counter clock is fixed at 30MHz to get maximum resolution. When D3 is operated in Tone mode, the same counter clock is changed to 1MHz and autoreload/ccr is updated to output variable PWM frequency on both D2 and D3 (20Hz to 20KHz).

So my advice is to use a different TIM pin say D2(TIM3) as PWM for backlight and D0/D1 as Tone.

wesner0019, 2015-07-29:
Thanks, that's what I did for a new run of my pcb's except I'm using laready using D0/D1 as I2C.

wesner0019, 2015-08-03:
@satishgn, I am on 4.0.3, and using just tone on A7. I am hearing it pulsing and not a solid tone. Any ideas? Below is the code I used.

```cpp
#define Buzzer A7 //Buzzer for sound notification

void setup() {
    pinMode(Buzzer, OUTPUT);    //Tone generator
    tone(Buzzer, 2150);
}

void loop(void) { 
}
```

satishgn, 2015-08-03:
@wesner0019, the tone fix is not released yet for the Web IDE. You will have to locally build using the "develop" branch.

wesner0019, 2015-08-03:
@satishgn, I built from the develop branch and the A7 pin is still pulsing when using tone. This pin is not related to the D2/D3 issue. From my understanding, the A7 tone pin should be working even without the develop branch.

paulmand3l, 2015-08-03:
Will this get analogWrite working on TX and RX as well?

satishgn, 2015-08-06:
@paulmand3l analogWrite should work on TX and RX using the develop branch.
@wesner0019 I do hear a pulsing sound when using A7 pin. looking into it now.

wesner0019, 2015-08-06:
@satishgn, thanks, any new findings?

satishgn, 2015-08-07:
@wesner0019, found that A7's TIM5 on STM32F2 chips is a 32-bit Timer whereas other TIM channel pin belongs to16 bit timers. That's a major difference. So writing 16-bit values to Autoreload and CCR's 32-bit registers is not behaving correct. I will have to read more on TIM5 in the STM32 reference manual and find out if it is a known issue. Will update once I find a solution to this problem.

satishgn, 2015-08-10:
@wesner0019 got the Tone issue fixed in the develop branch of firmware repo. please pull the latest and do a clean build of the system parts before flashing. Tone should now generate a solid Tone(no pulsing) on the A7 pin. Closing this issue.

closed 2015-08-10 by satishgn

wesner0019, 2015-08-10:
@satishgn THANKS

commit 7cb14bc1 (2015-07-24): Enable Tone generation on D2 and D3 - Fixes #483
commit b541fe40 (2015-07-24): Enable Tone generation on RX and TX - Fixes #483