particle-iot/device-os #484: SYSTEM_MODE(MANUAL) prevents WiFi/cloud connection on Core (develop only)

pomplesiegel, 2015-07-14:
Realized from debugging issue #482. 

When running simple SYSTEM_MODE(MANUAL) program, Core cannot successfully connect to WiFi/cloud.

Following program was compiled locally (using tip of develop).

```cpp
SYSTEM_MODE(MANUAL);

void setup()
{

  delay(3000);
  Serial.begin(9600); //Begin USB serial session
  WiFi.off(); 

  Wire.setSpeed(CLOCK_SPEED_100KHZ);
  Wire.stretchClock(true); //enable clock stretching
  Wire.begin();

  WiFi.on(); //Turn on Wifi chipset
  Spark.connect();

  Serial.println("Finished setup()");
}

void loop()
{
  Spark.process()
  delay(10);
}
```

At startup the core breathes blue, without successfully connecting to the cloud. Compiling using `Master` branch allowed the Core to instantly connect and breathe cyan. 

The only way I can get it breathing cyan is to place it into listening mode and successfully calling particle setup wifi.

I've tried

* Adding Hard-coded wifi creds in setup()
* Adding if(!Spark.connected()) Spark.connect(); to loop()
* Removing the 10ms delay in loop()
* Remvoing WiFi.off() setup()  

but to no avail. Even after getting it to breathing cyan, if I reset the spark, it goes back to breathing blue.

m-mcgowan, 2015-07-14:
Thanks for this. Does it improve if you remove the calls to `Wire.xxx()`? Doing that would make the example minimal (a [SSCCE](http://sscce.org))

pomplesiegel, 2015-07-14:
Good idea!

Just checked. w/ the calls to `Wire` commented out, the problem persists.

m-mcgowan, 2015-07-14:
I reproduced the issue. Here's an explanation for why it fails

- WiFi.on() turns on the wifi module, but does not connect to a network (breathing blue)
- Spark.connect() tells the system that it's ok to connect to the cloud if there is a wifi connection

So the system will connect as soon as there is a Wi-Fi connection. To fix the code and get the core online, change `WiFi.on()` to `WiFi.connect()`

pomplesiegel, 2015-07-14:
Thanks for this.

Previously in `Master` Spark.connect() called WiFi.connect() 

```cpp
void SparkClass::connect(void)
{
	//Schedule Spark's cloud connection and handshake
        WiFi.connect();
	SPARK_CLOUD_CONNECT = 1;
}
```
Will the new documentation specify that WiFi.connect() must now be called in addition?

m-mcgowan, 2015-07-14:
Ok,I see. The WiFi.connect() call was removed because it's a blocking call, and we want Spark.connect() to return as quickly as possible. 

I can look at setting some flags that will cause the system to start up the wifi connection.

pomplesiegel, 2015-07-14:
Sounds good, thanks!

closed 2015-08-22 by m-mcgowan