particle-iot/device-os #485: 0.4.3 release for Photon/P1 wrong location for P1 2b04:d006 should be 2b04:d008

wesner0019, 2015-07-14:
P1:

dfu-util -d 2b04:d006 -a 0 -s 0x8020000 -D system-part1-0.4.3-p1.bin
dfu-util -d 2b04:d006 -a 0 -s 0x8060000:leave -D system-part2-0.4.3-p1.bin
 
These should be:
dfu-util -d 2b04:d008 -a 0 -s 0x8020000 -D system-part1-0.4.3-p1.bin
dfu-util -d 2b04:d008 -a 0 -s 0x8060000:leave -D system-part2-0.4.3-p1.bin

m-mcgowan, 2015-07-14:
Thanks. fixed! (Copy paste error!)

closed 2015-07-14 by m-mcgowan

wesner0019, 2015-07-14:
just saw this too;
this location also needs updating
so devices can be returned to factory condition. To apply, download the file then run dfu-util -d 2b04:d006 -a 0 -s 0x8020000:leave -D unlocker-p1.bin

m-mcgowan, 2015-07-14:
You don't need to use unlocker when flashing via DFU.

m-mcgowan, 2015-07-14:
I see what you mean - the DFU ID was d006. Fixed. Thanks :-)