particle-iot/device-os #487: Can not download firmware using SWD if configuring the D7 as GPIO in the old firmware

XuGuohui, 2015-07-15:
Hi,
If the D7, i.e. JTMS pin is configured as GPIO to control the on-board LED, when I want to download a new firmware using SWD via JLink or OpenOCD, it can not detect the target and abort. How can fix it? Any idea would be appreciated!

Best regards

m-mcgowan, 2015-07-15:
Hi @XuGuohui - I've never tried this or been successful, but you might try setting the pinMode to input pullup or pull down.

Alternatively, if you put the core in DFU mode, the JTAG pins will still be in their default correct state so you can then successfully flash new firmware. 

I hope that helps!
mat.

XuGuohui, 2015-07-15:
Hi mat.

Thanks for your suggestion!
As I'm using the SWD to program the whole flash, so there is no DFU mode in application flash, except that running from the ST system bootloader, but it has to change the hardware to modify the boot configuration.

technobly, 2015-07-20:
@XuGuohui if your code is using D7 as a GPIO, and you are not initially compiling your firmware with the USE_SWD_JTAG=y option, D7 will prevent the JTAG tools from attaching to the device.  The quick fix is to hold RESET on the Photon/Core when trying to attach your JTAG tool.  If reset is not an acceptable fix, you can use SWD mode and require less pins than JTAG.  To do this compile your code with the USE_SWD=y option.  I'm pretty sure this will free up the D3 and D5 pins, to which you could potentially move the D7 function to one of those.  Some more digging will be necessary though to be sure. Here are some links:

https://www.segger.com/interface-description.html

http://docs.particle.io/photon/photon-datasheet/#2-5-jtag

XuGuohui, 2015-07-24:
@technobly Thanks for your help. After looking into this issue for a long time, I think the key point is to issue a reset signal by JTAG/SWD emulator automatically when it can't detect the target MCU. So maybe this is a bug at JTAG/SWD emulator side. Although the methods you recommended and I have searched work to resovle this issue, it's the responsibility on JTAG/SWD emulator to fix this issue, since I found certain tools can detect the target MCU event I configure the JTAG pins as GPIO in the old firmware. I suppose that if the emulator issue a reset signal before its following action will resolve this issue.

technobly, 2015-07-24:
@XuGuohui yes you are correct, and I know in the ST Link Utility for Windows, there is a Connect Under Reset.  Not sure if this is an option built into the texane/stlink, there are reports that st-flash --reset does not work, also I have apparently already asked this question over there with no response: https://github.com/texane/stlink/issues/220

technobly, 2015-07-24:
It looks like `st-flash --reset write somebinary.bin 0x8000000` with the texane/stlink may be the way to do this... looks like it's hooked up in code also:
```
    // checking for options
    if (strcmp(av[0], "--reset") == 0)
    {
        o->reset = 1;
        ac--;
        av++;
    }
```
and 
```
    // using options
    if (o.reset){
        stlink_jtag_reset(sl,2);
        stlink_reset(sl);
    }
```

XuGuohui, 2015-07-28:
@technobly I used to use the ST Link Utility to detect the target MCU under Connect Under Reset mode and it did work even if I configured the JTAG/SWD pins as GPIO in the old firmware. And I also found that the JFlashARM utility built in the JLink can erase the target MCU in the same situation, but only limited in erase operation and other operations like downloading firmware are still failed detecting target MCU. Here I list the methods those I have found so far to fix this issue:
1. Change the boot configuration(Please refer to the ST's documents) to make ST MCU boot from system bootloader or RAM, since these mode will keep JTAG/SWD active.
2. Try pressing the reset button on your board and release it once the JTAG/SWD emulator tries to connect to the target MCU. It depends on your luck since it will fail if you release the reset button not in proper timer.
3. Use the ST Link emulator cooperate with the ST Link Utility tool to download your new firmware
4. Use Jlink cooperate with JFlashARM utility to erase the old firmware and then program the new firmware.
5. Make a bootloader and in the bootloader the JTAG/SWD is active, thus you can enter the bootloader and then use JTAG/SWD emulator to download your new firmware.

Any other methods are welcomed!

In a word, ensure the JTAG/SWD interface of the target MCU is active when emulator try to connect to it.

closed 2015-07-28 by XuGuohui