particle-iot/device-os #488: Missing HAL_disable_irq and HAL_disable_irq in dynalib interface

maghis, 2015-07-15:
This took a while to figure out...

m-mcgowan, 2015-07-15:
Hi @maghis! Sorry it took you a while to figure this out. What's your use case for these functions? Disabling interrupts on a multi-threaded system can lead to some issues so we'd prefer to meet your needs some other way if possible. Please let me know what you're trying to achieve and we'll work out a solution together! 
Cheers,
mat.

maghis, 2015-07-15:
Hey man, no worries.

I'm porting old code over and it's disabling interrupts in very specific parts, I'll move it to mutexes soon.
Since we have the HAL I didn't want to include arch specific headers and stick with it.
It's also nice that we get the mask out, nice api.

m-mcgowan, 2015-07-15:
I've been thinking about providing atomic blocks so that we abstract how to achieve that. (Disabling interrupts works, but it's using a hammer to crack an egg!)  Something similar to the atomic macros available on the AVR platform - http://www.nongnu.org/avr-libc/user-manual/group__util__atomic.html#ga362c18b15a09703e42e1c246c47420ef

maghis, 2015-07-15:
Well, now arm gcc has built-in support for CAS operations.
For example you could do something like:
bool busy = false; <- somewhere

// acquire lock
do
{
  // some kind of spin wait or maybe delay to release the thread?
} while  (__sync_bool_compare_and_swap(&busy, false, true)); // till the CAS succeeds

You could also include a timeout to acquire the lock (a very good practice).
The api could look something like:

bool try_acquire_lock(bool* sync, int timeout);
void release_lock(bool* sync);

maghis, 2015-07-15:
I don't like misterious macros too much, but you could have a macro that generates the calls to acquire_lock and release_lock.

The above implementation can go fancier and fancier, for example you could add support for nested locks on the same sync, etc etc...

m-mcgowan, 2015-07-15:
This is good if the problem you're solving is concurrent access to a resource. Is that why you need to disable interrupts - to serialize access to a shared resource?

maghis, 2015-07-15:
Yup, the i2c periph.

We have a timer based sampling of some i2c devices (about 200Hz) and there are other i2c devices that we need to access very sporadically (once every few seconds).
I have a super low i2c timeout 500us and I disable interrupts during the sporadic sampling.

maghis, 2015-07-15:
@m-mcgowan talking about that, I have to use a timer interrupt, do you think the best way is to expand the hal to support them?

m-mcgowan, 2015-08-11:
fixed in this commit - https://github.com/spark/firmware/commit/185c4c0fd228683cd8531e43814bfb9df9c2ccd8

closed 2015-08-11 by m-mcgowan